riscv-software-src/riscv-tools #15: Build issues on Mac OS X 10.11 (El Capitan)

tommythorn, 2016-01-01:
It's not clear how to build this on a Mac (it has worked in the past):

```
2016-01-01 12:40:35 ~/riscv-tools
git pull; git clean -fdxq; git reset --hard master
Already up-to-date.
HEAD is now at 5ddcdfd want to clone linux, not tools
2016-01-01 12:41:08 ~/riscv-tools
export CC=gcc-4.8;export CXX=g++-4.8;export RISCV=/opt/rv
2016-01-01 12:41:42 ~/riscv-tools
./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-fesvr/build directory
Configuring project riscv-fesvr
configure: error: in `/Users/tommy/riscv-tools/riscv-fesvr/build':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
```

alansuphd, 2016-01-02:
I am trying to build on El Capitan too. I have problem installing gcc4.8 from MacPort so I installed gcc49 instead to compile the rocket chip simulator. While building the tool chain I run into this error:

100  116M  100  116M    0     0   267k      0  0:07:27  0:07:27 --:--:--  247k
./contrib/download_prerequisites: line 32: wget: command not found
make: *** [src/gcc] Error 1

Does upgrade to gcc5 helps?

tommythorn, 2016-01-02:
Try 
$ brew install wget

There are a couple of dependencies.  I think some are (all?) are mentioned in
the README.md but otherwise we should add them.

Tommy

alansuphd, 2016-01-02:
Thanks a lot! This issue solved! It is moving forward again.

I do not think README mentions this one, nor the need for GNU sed and awk. I stopped twice to installed them then came the wget.

Anyway, thanks again!

alansuphd, 2016-01-02:
Just to report the build is successful on Mac OS X 10.11.1 (El Capitan) with gcc 4.9.3.

I will try gcc5 too later.

kionix, 2016-01-02:
hi,
just to chip in my experience, I've had no issues building (and running)
the riscv toolchain on El Capitan using Apple's clang that comes with xcode
(i guess commandline tools would do as well). The other dependencies came
from homebrew. Just don't install mpfr and gmp libs so the build system
downloads the proper versions itself ( u could also set the appropriate
configure switches).

Kamyar

alansuphd, 2016-01-04:
Thanks kionix!

Out of interest I change using Apple's clang to compile risk-tools, the RISC-V toolchain, with
CC=cc CXX=c++ ./build.sh

I double check to make sure cc and c++ are indeed the Apple clang installed at /usr/bin and use -v to make sure they are indeed Apple LLVM version 7.0.2 (clang-700.1.81).

After riscv-fester has been built and into the riscv-isa-sim build, the compilation errors out saying:

Configuring project riscv-isa-sim
Building project riscv-isa-sim
clang: warning: treating 'c-header' input as 'c++-header' when in C++ mode, this behavior is deprecated
In file included from ../riscv/insn_template.h:3:
In file included from ../riscv/mmu.h:10:
../riscv/processor.h:119:15: error: implicit instantiation of undefined template
      'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
  std::string isa;
              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
1 error generated.
make: *** [insn_template.h.gch] Error 1

Checking into the code and cannot find problems there. For a quick verification to identify if this a compiler problem, at the exact same source tree I change the compiler to gcc4.9 then the riscv-isa-sim build successfully. Since my interest is to study RISC-V so, I will not pursue further with the Apple clang unless someone kind enough to address mistakes I make in using Apple clang.

Alan

sbeamer, 2016-01-04:
If you go into riscv-tools and check out the head of master for riscv-isa-sim there is already a fix for this bug.

In general, we periodically make sure riscv-tools can compile on OS X just with using clang. I'm currently try to sort out the latest hiccups in time for the workshop.

sbeamer, 2016-01-05:
So the main compile bugs people have been having (including our homebrew's travis) was fixed by a recent https://github.com/riscv/riscv-gnu-toolchain/pull/112. It should fully propagate into riscv-tools and homebrew-riscv in a bit.

alansuphd, 2016-01-06:
Thanks and hope you guys have fund in the workshop! But please bear with me for a stupid question... how exactly I check out the latest updates? In riscv-tools I tried 'git submodule update --init --recursive' and nothing happened. I am not sure if latest updates are checked out.

colinschmidt, 2016-01-06:
You need to 'git pull' in riscv-tools to get the new submodule hashes and then 'git submodule update --init --recursive'  will checkout the submodules at those hashes.

alansuphd, 2016-01-07:
Hi Colin,

thanks! I had successfully updated the source tree and had compiled in clang.

Cheers!

Alan

__________________________
Alan P. Su, Ph.D
alansuphd@gmail.com
(C) +886-928883254

commit b50e0752 (2016-01-11): Point to Homebrew for a gcc, fixes #15
commit d30a479e (2016-01-11): Merge pull request #16 from tommythorn/issue-15

Point to Homebrew for a gcc, fixes #15
closed 2016-01-11 by sbeamer