riscv-software-src/riscv-tools #19: Docker container for riscv-tools

sbates130272, 2016-01-05:
Hi 

To help people get going with RISC-V I have containerized most of the stuff in the riscv-tools repo. The container image can be pulled from https://hub.docker.com/r/sbates130272/riscv/ and the underlying dockerfile can be found at https://github.com/sbates130272/docker-riscv.

It would be great to have people do a bit of testing on this before emailing the wider audience via the mailing list. The plan would be to update this container when new "releases" of the riscv-tools repo is done.

Note there are currently a couple of issues. One is that "mount" is not allowed in a Docker contain and the other is you need AUFS support as the image size exceeds 10G. I'd be happy to review PRs against the Dockerfile is people have better options for these issues or want to add more things to the image.

Stephen

ryepesg, 2017-05-30:
Great idea Stephen.
 
In the riscv-qemu project they suggest:

> To avoid having to build the RISC-V toolchain and programs yourself, use Stefan O'Rear's RISC-V Fedora Docker Image to obtain a Fedora 25 Userland for RISC-V, packaged with riscv-qemu.

The link is: https://hub.docker.com/r/sorear/fedora-riscv-wip

What I like is the small size (130 MB).

Cheers,

sorear, 2017-05-30:
They aren't at all comparable.  One of them is a cross-toolchain (a bunch of amd64 binaries), the other is a base OS image (a bunch of riscv64 binaries).

ryepesg, 2017-05-30:
You are right, they are different, but I think the base OS image also meets the desire of Stephen of "help people get going with RISC-V". I think both are useful in different situations and both could be mentioned in the main page.

The fedora docker image is the fastest way I know for get people to play with RISC-V, no need to download GB of files or wait for hours for the riscv-tools to get compiled (for each of my students' machines!). With your fedora image, I just code some quick in C an then I use gcc to show the assembly code (gcc -S file.c) or I just use the assembler and linker with gcc all in one step (gcc file.s).