riscv-software-src/riscv-tools #20: Unrecognized opcode 'pushargs_' & 'popargs_'

alansuphd, 2016-01-12:
I am on Mac OS X El Capitan building riscv-tools. With Clang version:
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.0.0
Thread model: posix

I have successfully built the tool chain and fesvr-eth, spike, riscv64-unknown-elf-* and such are all created. Then move next to the riscv64-unknown-linux-gnu-gcc build. After long into the build with most of the riscv64-unknown-linux-gnu-* done I got an error in the posix build. It stopped at pread.c compilation as follows:

(echo '#define SYSCALL_NAME pread'; \
	 echo '#define SYSCALL_NARGS '; \
	 echo '#define SYSCALL_SYMBOL __libc_pread'; \
	 echo '#define SYSCALL_CANCELLABLE 1'; \
	 echo '#define SYSCALL_NOERRNO 0'; \
	 echo '#define SYSCALL_ERRVAL 0'; \
	 echo '#include <syscall-template.S>'; \
	 echo 'weak_alias (__libc_pread, __libc_pread64)'; \
	 echo 'hidden_weak (__libc_pread64)'; \
	 echo 'weak_alias (__libc_pread, __pread)'; \
	 echo 'hidden_weak (__pread)'; \
	 echo 'weak_alias (__libc_pread, pread)'; \
	 echo 'hidden_weak (pread)'; \
	 echo 'weak_alias (__libc_pread, __pread64)'; \
	 echo 'hidden_weak (__pread64)'; \
	 echo 'weak_alias (__libc_pread, pread64)'; \
	 echo 'hidden_weak (pread64)'; \
	) | riscv64-unknown-linux-gnu-gcc -m64 -c     -I../include -I/Users/phdasu/RD/OddPoint/RISC-V/rocket-chip-Clang/riscv-tools/riscv-gnu-toolchain/build-glibc-linux64/posix  -I/Users/phdasu/RD/OddPoint/RISC-V/rocket-chip-Clang/riscv-tools/riscv-gnu-toolchain/build-glibc-linux64  -I../sysdeps/unix/sysv/linux/riscv/rv64  -I../sysdeps/unix/sysv/linux/riscv  -I../sysdeps/riscv/nptl  -I../sysdeps/unix/sysv/linux/generic  -I../sysdeps/unix/sysv/linux/wordsize-64  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/riscv/rv64  -I../sysdeps/ieee754/flt-32  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/riscv/rv64/soft-fp  -I../sysdeps/riscv  -I../sysdeps/init_array  -I../sysdeps/wordsize-64  -I../sysdeps/riscv/fpu  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I.   -D_LIBC_REENTRANT -include /Users/phdasu/RD/OddPoint/RISC-V/rocket-chip-Clang/riscv-tools/riscv-gnu-toolchain/build-glibc-linux64/libc-modules.h -DMODULE_NAME=libc -include ../include/libc-symbols.h       -DASSEMBLER  -g -Werror=undef    -o /Users/phdasu/RD/OddPoint/RISC-V/rocket-chip-Clang/riscv-tools/riscv-gnu-toolchain/build-glibc-linux64/posix/pread.o -x assembler-with-cpp - -MD -MP -MF /Users/phdasu/RD/OddPoint/RISC-V/rocket-chip-Clang/riscv-tools/riscv-gnu-toolchain/build-glibc-linux64/posix/pread.o.dt -MT /Users/phdasu/RD/OddPoint/RISC-V/rocket-chip-Clang/riscv-tools/riscv-gnu-toolchain/build-glibc-linux64/posix/pread.o

and came the error message:

../sysdeps/unix/syscall-template.S: Assembler messages:
../sysdeps/unix/syscall-template.S:84: Error: unrecognized opcode `pushargs_'
../sysdeps/unix/syscall-template.S:84: Error: unrecognized opcode `popargs_'

Anyone can help? I tried looking for past issues and cannot find this one. I am still new to github so please bear with me if I am wrong.

Thanks!

sbeamer, 2016-01-12:
Were you building riscv64-unknown-linux-gnu-* within a case-sensitive file system (not default on OS X)? We recommend building riscv64-unknown-linux-gnu-* within a disk image with a case sensitive format.

sbeamer, 2016-01-12:
Additionally, to build glibc you will probably need to use gcc.

aswaterman, 2016-01-12:
Sounds like gsed or gawk is not installed.  Check on that first.

alansuphd, 2016-01-13:
Thanks to all the comments!

Well, gsed and gawk are installed. Just to be sure, the gsed version is 4.2.2 and gawk 4.1.3. However there also BSD sed and awk in the system, they come with OS X. Is this a problem?

OS X is indeed case insensitive. And it requires a huge effort, to create a new partition and format it into case sensitive. I do not like to play around with iMac like that. This iMac is the only working environment I have. Do not have the resource to build a PC Linux.

Any other thoughts? Maybe hack into the code and make it work? How?

BTW, the riscv-tools/README.md says use apt-get to install all tools needed. Well, apt-get is not supported on iMac as far as I know. Care to modify the README.md?

alansuphd, 2016-01-13:
Oh, yea, I followed the procedure on README to install linux tool chain. After  "./configure --prefix=$RISCV" it is just to "make linux". I believe it uses gcc (comes with Xcode) instead of Clang cc.

tommythorn, 2016-01-13:
Huge effort? Just create a file system in a file and mount that!

Tommy

alansuphd, 2016-01-13:
Ask the wrong question on the web and get the wrong answer. Several comments against the idea to change the case sensitivity on earlier OS X. And then it turns out the El Capitan's Disk Utility can resize current partition to make room for a new case sensitive partition to be created. Oh well, keep going on and hope this time it works...

alansuphd, 2016-01-13:
Oops... case sensitive does not help. The build stops at exactly the same place with exactly the same error message. Any other thoughts?

Mind that this is already very deep into the build. In riscv/bin there are 24 riscv64-unknown-linux-gnu-* entries successfully built as follows:

riscv64-unknown-linux-gnu-addr2line
riscv64-unknown-linux-gnu-ar
riscv64-unknown-linux-gnu-as
riscv64-unknown-linux-gnu-c++filt
riscv64-unknown-linux-gnu-cpp
riscv64-unknown-linux-gnu-elfedit
riscv64-unknown-linux-gnu-gcc
riscv64-unknown-linux-gnu-gcc-5.3.0
riscv64-unknown-linux-gnu-gcc-ar
riscv64-unknown-linux-gnu-gcc-nm
riscv64-unknown-linux-gnu-gcc-ranlib
riscv64-unknown-linux-gnu-gcov
riscv64-unknown-linux-gnu-gcov-tool
riscv64-unknown-linux-gnu-gprof
riscv64-unknown-linux-gnu-ld
riscv64-unknown-linux-gnu-ld.bfd
riscv64-unknown-linux-gnu-nm
riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-objdump
riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-strip

sbeamer, 2016-01-13:
For making the volume, you can just make a disk image (don't need to make a partition). You can do it with Disk Utility (https://support.apple.com/kb/PH22247?viewlocale=en_US&locale=en_US).

Yes, it is OK for the BSD variants to coexist with the GNU versions of sed and awk.

Did you try building glibc with gcc? If you don't have gcc installed, I would install it from Homebrew.

alansuphd, 2016-01-14:
Since Clang seems not live up to my expectations, I like to migrate to the latest and yet stable gcc, and I can jump to it directly to eliminate the complication of migrating from version to version, a mistake I made earlier that messed up the compiling environment. Can you recommend a gcc version?

Thanks a lot!

sbeamer, 2016-01-14:
Any gcc newer than 4.8 will do. In general, I haven't seen many issues with changing gcc versions.

For OS X, the easiest way to get gcc is with [Homebrew](http://brew.sh) and running `brew install gcc`. Homebrew's current default gcc is 5.3.

alansuphd, 2016-01-15:
Ok, thanks Scott!

Before I go further, please let me dig deeper into this problem. I put a lot of thoughts to ask the right question. To the end, let me ask this: using El Capitan default gcc, how do I build using glibc?

The gcc library in El Capital has /usr/lib/libc.dylib. But I cannot find a way to check if it complies to glibc. Then, please be noted that in this error the compiler is riscv64-unknown-linux-gnu-gcc, the one that built out of earlier and the same Linux gcc build. Therefore it should be using glibc. However in riscv/lib I cannot find libc.a, libc.so.6 or anything named with libc (please see the list below). I just want to be clear what is the problem here and make my next move knowing what I am doing. I believe your suggestion in using glibc a good one, so the question above.

Libraries I have in riscv/lib are:

libcc1.0.so
libcc1.la
libcc1.so -> libcc1.0.so
libdummy_rocc.so
libfesvr.so
libhwacha.so
libriscv.so
libsoftfloat.so
libspike_main.so

sbeamer, 2016-01-15:
Default gcc on OS X is actually just mapped to clang. If you install gcc (5.3 by default) with Homebrew, you should be good to go. To use the gcc installed by Homebrew, you should override the environments variables

    $> CC=gcc-5 CXX=g++-5 make

alansuphd, 2016-01-16:
gcc-5 installed successfully, and riscv-tools build also successfully, then into the Linux gcc build. Unfortunately again after very deep into the build and 24 riscv64-unknown-linux-gnu* entries are built, run into yet an old problem: not the correct ld and as. Error message attached at the back.

Using brew to install binutils and yet still no gnu ld and as in sight. Cannot find a solution on the web. Some old comments even say gnu ld and as are not supported on Mac. Any suggestions?

Error messages:

checking build system type... x86_64-apple-darwin15.0.0
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... gcc-5
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc-5 accepts -g... yes
checking for gcc... gcc
checking for riscv64-unknown-linux-gnu-readelf... riscv64-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... yes
checking whether g++-5 accepts -g... yes
checking whether g++-5 can link programs... yes
checking for sysdeps preconfigure fragments... aarch64 alpha arm hppa i386 m68k microblaze mips nacl nios2 powerpc riscv s390 sh sparc tile x86_64 
configure: running configure fragment for add-on libidn
checking for assembler and linker STT_GNU_IFUNC support... no
checking whether .text pseudo-op must be used... yes
checking sysdep dirs... sysdeps/unix/sysv/linux/riscv/rv64 sysdeps/unix/sysv/linux/riscv sysdeps/riscv/nptl sysdeps/unix/sysv/linux/generic sysdeps/unix/sysv/linux/wordsize-64 sysdeps/unix/sysv/linux sysdeps/nptl sysdeps/pthread sysdeps/gnu sysdeps/unix/inet sysdeps/unix/sysv sysdeps/unix sysdeps/posix sysdeps/riscv/rv64 sysdeps/ieee754/flt-32 sysdeps/ieee754/dbl-64 sysdeps/riscv/rv64/soft-fp sysdeps/riscv sysdeps/init_array sysdeps/wordsize-64 sysdeps/riscv/fpu sysdeps/ieee754 sysdeps/generic
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln -s works... yes
checking whether as is GNU as... no
checking whether ld is GNU ld... no
checking for as... as
checking version of as... v. ?.??, bad
checking for ld... ld
checking version of ld... v. ?.??, bad
checking for gnumake... gnumake
checking version of gnumake... 3.81, ok
checking for gnumsgfmt... no
checking for gmsgfmt... no
checking for msgfmt... msgfmt
checking version of msgfmt... 0.19.7, ok
checking for makeinfo... makeinfo
checking version of makeinfo... 4.8, ok
checking for sed... /opt/local/bin/gsed
checking version of /opt/local/bin/gsed... 4.2.2, ok
checking for gawk... /opt/local/bin/gawk
checking version of /opt/local/bin/gawk... 4.1.3, ok
checking if gcc-5 is sufficient to build libc... yes
checking for riscv64-unknown-linux-gnu-nm... riscv64-unknown-linux-gnu-nm
configure: error: 
*** These critical programs are missing or too old: as ld
*** Check the INSTALL file for required versions.
make: *** [stamps/build-glibc-linux64] Error 1