riscv-software-src/riscv-tools #28: build.sh: GDB not automatically installed?

neuschaefer, 2016-06-04:
Hi,

I was trying to integrate the spike test suite into the riscv-tools `.travis.yml`, but I noticed that gdb wasn't installed ([build log][1]). I also noticed that the riscv-gnu-toolchain Makefile.in uses `--disable-gdb` in a few places. What's the status of testing gdb in travis?

[1]: https://travis-ci.org/neuschaefer/riscv-tools/builds/135219360

palmer-dabbelt, 2016-06-04:
Up until a few days ago, GDB didn't build with the standard RISC-V binutils releases (it was ported on another branch).  That's why the builds are disabled.  I'd accept a patch turning it on, now that it should actually work.

neuschaefer, 2016-06-05:
I'll see if I can write a patch.

timsifive, 2016-06-16:
Just want to add that I'd love for this to be fixed. gdb wasn't building for me, and I suspect it's because people just run "build.sh" and if that passes, they think their changes are good.

palmer-dabbelt, 2017-08-26:
This should be fixed.

closed 2017-08-26 by palmer-dabbelt