riscv-software-src/riscv-tools #41: 32b simulation issue

morse-michael, 2016-09-09:
Hi all, 

I am able to compile & run riscv-tools in 64b mode but I am having some trouble with getting 32b applications running. I seem to be able to compile riscv-tools for 32b mode but riscv-tests failed to compilete, see my comment on https://github.com/riscv/riscv-tools/pull/39 for more informations.

When I try to run the helloworld application as documented in the README file I get the expected behavior in my 64b environment, but it fails in my 32b environment, see below.

What I have:
- 32b installation of HoB riscv-tools in  /tools/riscv-tools/install/riscv-tools.20160908-master-32b
- 64b installation of HoB riscv-tools in  /tools/riscv-tools/install/riscv-tools.20160908-master-64b

64b env:

> VER=20160908-master-64b
> export PATH=$PATH:/tools/riscv-tools/install/riscv-tools.$VER/bin/
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/tools/riscv-tools/install/riscv-tools.$VER/lib64/:/tools/riscv-tools/install/riscv-tools.$VER/lib/
> riscv64-unknown-elf-gcc -o hello64 hello64.c
> spike pk hello64
> warning: only got 2931818496 bytes of target mem (wanted 4294967296)
> Hello world!

32b env:

> VER=20160908-master-32b
> export PATH=$PATH:/tools/riscv-tools/install/riscv-tools.$VER/bin/
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/tools/riscv-tools/install/riscv-tools.$VER/lib64/:/tools/riscv-tools/install/riscv-tools.$VER/lib/
> riscv32-unknown-elf-gcc -o hello32 hello32.c
> spike pk hello32
> warning: only got 2931818496 bytes of target mem (wanted 4294967296)
> terminate called after throwing an instance of 'std::runtime_error'
>  what():  could not open pk

When I point it to the 32b version of pk I no longer get the "could not open" error above but I just get this (no Hello World!):

> spike /tools/riscv-tools/install/riscv-tools.20160908-master-32b/riscv32-unknown-elf/bin/pk hello32
> warning: only got 2931818496 bytes of target mem (wanted 4294967296)

Anyone any idea of what could be wrong?

Thanks,

Michael

morse-michael, 2016-09-09:
Oh, found it, looks like I need to pass the ISA string of a 32b architecture.
The following command works for me:

> spike --isa=RV32IMAFDC /tools/riscv-tools/install/riscv-tools.20160908-master-32b/riscv32-unknown-elf/bin/pk hello32
> warning: only got 2931818496 bytes of target mem (wanted 4294967296)
> Hello world!

Would be good to add this to the README :-).

Note that I still need to provide the full path to the 32b pk, spike can't find it otherwise. Also build_project riscv-tests still fails in 32b mode (see my comment on https://github.com/riscv/riscv-tools/pull/39).

morse-michael, 2016-09-09:
Looks like I have to set the following in build.sh to get riscv-tests passing:

> export XLEN=32

shrshore, 2016-11-23:
my os version is 
```
Linux ubuntu32 4.4.0-31-generic #50~14.04.1-Ubuntu SMP Wed Jul 13 01:06:37 UTC 2016 i686 i686 i686 GNU/Linux
```
and when i try the Building riscv64-unknown-linux-gnu-gcc (11.41 SBU):
```
#ubuntu32:~/riscv-tools/riscv-gnu-toolchain$ ./configure --prefix=$RISCV --with-xlen=32
#ubuntu32:~/riscv-tools/riscv-gnu-toolchain$ make linux
```
The compiler notifies me that the __int128 is not supported for this target,
```
/home/shore/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/sim-main.c: In function ‘mulhu’:
/home/shore/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/sim-main.c:602:12: error: ‘__int128’ is not supported for t                                  his target
   return ((__int128)a * b) >> 64;
```
I have no idea how to fix it, could you help me? Thanks a ton....

AngelTerrones, 2016-11-23:
@shrshore the problem is that you are using a 32bits OS. The solution is described here: https://groups.google.com/a/groups.riscv.org/forum/#!msg/sw-dev/Ejcaqshly3o/KhP_qaSbHQAJ

shrshore, 2016-11-24:
@AngelTerrones Wow, Thanks so much...I had searched hard but found nothing helpful..orz
Now I can build the project and get riscv32-unknown-linux-gnu tools.  

By the way, do you know when i use riscv32-unknown-linux-gnu-gcc to build a linux kernel(the project is [here](https://github.com/riscv/riscv-linux) and I have checkout to priv-1.9 branch), how to change the target executable file to elfriscv32 instead of elfriscv64?
because when I command
```
make ARCH=riscv vmlinux
```
it remind me that
```
Fatal error: selected target format 'elf64-littleriscv' unknown
```
I have modified general setup->Cross-compiler tool prefix to  riscv32-unknown-linux-gnu- in menuconfig.

I will appreciate it very much if you can help me once again!

AngelTerrones, 2016-11-25:
@shrshore Sorry, no idea. BTW, I think you should ask that question in the mailing list, in order to keep this thread on-topic.

shrshore, 2016-11-26:
@AngelTerrones okay, thanks.

jim-wilson, 2018-02-03:
It appears that all of the questions were answered already.

closed 2018-02-03 by jim-wilson