riscv-software-src/riscv-tools #42: 32bit simulation error on isa RV32I

songchengwei, 2016-09-12:
Hi,  I download the riscv-tools, install the tool with --with-xlen=32 option, I can successfully run hello world with [https://github.com/riscv/riscv-tools/issues/41](issue 41), then I try to compile C code with the specific RV32I ISA, I tried two ways:

1, I try to use compile option -march=RV32I to compile the C code, and then use command:  spike --isa=RV32I /path/to/32bit/pk/pk hello. There is no output, but if I change the command to spike --isa=RV32, the output is as expected.
2, I try to install the tool-chain with option --with-arch=I --with-xlen=32, the result is as 1.

Is there anything wrong? How can I only use RV32I isa?

mehnadnerd, 2017-08-23:
I ran into some similar issues. I have found that the M and A extensions appear to be required in order to use C functions like printf (or puts, which it optimises it to). I am not sure how to resolve this issue.

palmer-dabbelt, 2017-08-23:
You probably have a C library that was built for rv32ima.  If you build one for rv32i then you shouldn't have any M or A instructions.

mehnadnerd, 2017-08-24:
I've found the instructions for how to do so here: https://github.com/ucb-bar/riscv-sodor#building-a-rv32i-toolchain.

jim-wilson, 2018-02-03:
Already answered.

closed 2018-02-03 by jim-wilson