riscv-software-src/riscv-tools #43: Having trouble building

mark-i-m, 2016-09-17:
I am getting a compile error when building with `./build.sh`.

I am on Ubuntu 16.04. I have tried gcc-4.8 and gcc-5.4.

I have installed the apt-gets in the README. I have also set RISCV and added $RISCV/bin to PATH.

```
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/_default_fcntl.h:184:0,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/fcntl.h:3,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/fcntl.h:1,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/search/hash_page.c:62:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/search/hash.c:44:0:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
make[6]: *** [lib_a-hash_page.o] Error 1
make[6]: *** Waiting for unfinished jobs....
make[6]: *** [lib_a-hash.o] Error 1
Makefile:1922: warning: overriding recipe for target 'lib_a-vfwscanf.o'
Makefile:1624: warning: ignoring old recipe for target 'lib_a-vfwscanf.o'
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/_default_fcntl.h:184:0,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/fcntl.h:3,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/stdio/fdopen.c:60:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
make[6]: *** [lib_a-fdopen.o] Error 1
make[6]: *** Waiting for unfinished jobs....
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/locale/ldpart.c:30:0:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
make[6]: *** [lib_a-ldpart.o] Error 1
make[6]: *** Waiting for unfinished jobs....
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/_default_fcntl.h:184:0,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/fcntl.h:3,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/fcntl.h:1,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/reent/fcntlr.c:5:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/reent/fstatr.c:6:0:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
make[6]: *** [lib_a-fcntlr.o] Error 1
make[6]: *** Waiting for unfinished jobs....
make[6]: *** [lib_a-fstatr.o] Error 1
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/reent/renamer.c:6:0:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/reent/mkdirr.c:47:0:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
make[6]: *** [lib_a-mkdirr.o] Error 1
make[6]: *** [lib_a-renamer.o] Error 1
In file included from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/_default_fcntl.h:184:0,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/fcntl.h:3,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/fcntl.h:1,
                 from /path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/reent/openr.c:5:
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:44:3: error: unknown type name 'blksize_t'
   blksize_t st_blksize;
   ^~~~~~~~~
/path/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/libc/include/sys/stat.h:46:3: error: unknown type name 'blkcnt_t'
   blkcnt_t st_blocks;
   ^~~~~~~~
make[6]: *** [lib_a-openr.o] Error 1
/bin/bash: line 17: cd: riscv: No such file or directory
cp: cannot stat 'riscv/lib.a': No such file or directory
make[7]: *** [lib.a] Error 1
make[6]: *** [all-recursive] Error 1
/path/rocket-chip/riscv-tools/riscv64-unknown-elf/bin/ar: ../search/lib.a: No such file or directory
/path/rocket-chip/riscv-tools/riscv64-unknown-elf/bin/ar: ../stdio/lib.a: No such file or directory
/path/rocket-chip/riscv-tools/riscv64-unknown-elf/bin/ar: ../locale/lib.a: No such file or directory
/path/rocket-chip/riscv-tools/riscv64-unknown-elf/bin/ar: ../reent/lib.a: No such file or directory
/path/rocket-chip/riscv-tools/riscv64-unknown-elf/bin/ar: ../machine/lib.a: No such file or directory
make[5]: *** [all-recursive] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all-target-newlib] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib] Error 2

```

mark-i-m, 2016-09-22:
A quick update: I was able to build in an Ubuntu 14.04 VM. I believe the error has to do with GDB failing to build.

MatthewMyunghaKim, 2016-11-03:
The same problem was happened to me on the Ubuntu 14.04. It had no problem before.

palmer-dabbelt, 2016-11-03:
This doesn't look like a RISC-V specific thing, and might have been fixed upstream: https://sourceware.org/ml/newlib/2016/msg00655.html .  We're pretty far away from the newlib tip so I think just trying to build the x86 port of the newlib version we use might reveal what's going on.  If that patch fixes the problem then I'm OK taking it into our newlib port, since we probably won't update to upstream's tip soon.

palmer-dabbelt, 2017-08-26:
I'm closing this, as I don't think it's RISC-V specific and it's quite old.

closed 2017-08-26 by palmer-dabbelt