riscv-software-src/riscv-tools #51: build-rv32ima.sh fails to build

timsifive, 2016-12-13:
First problem I ran into is that it uses the upper-case march format. Fixed that like this:
```
--- a/build-rv32ima.sh
+++ b/build-rv32ima.sh
@@ -8,8 +8,8 @@
 echo "Starting RISC-V Toolchain build process"
 
 build_project riscv-fesvr --prefix=$RISCV
-build_project riscv-isa-sim --prefix=$RISCV --with-fesvr=$RISCV --with-isa=RV32IMA
-build_project riscv-gnu-toolchain --prefix=$RISCV --with-arch=RV32IMA
-CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv32-unknown-elf
+build_project riscv-isa-sim --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv32ima
+build_project riscv-gnu-toolchain --prefix=$RISCV --with-arch=rv32ima
+cC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv32-unknown-elf
 
 echo -e "\\nRISC-V Toolchain installation completed!"
```

Then I got the following build error:
```
...
/usr/bin/install: cannot stat ‘/home/tnewsome/SiFive/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/po/zh_CN.gmo’: No such file or directory
*** This configuration is not supported in the following subdirectories:
     zlib target-libquadmath target-libatomic target-libcilkrts target-libitm target-libsanitizer target-libvtv target-libmpx gnattools gotools target-libada target-libgfortran target-libbacktrace target-libgo target-libffi target-zlib target-libjava target-libobjc target-libgomp target-liboffloadmic target-libssp target-boehm-gc
    (Any other directories should still work fine.)
configure: WARNING: decimal float is not supported for this target, ignored
libtool: install: warning: remember to run `libtool --finish /opt/riscv/libexec/gcc/riscv32-unknown-elf/6.1.0'
configure: WARNING: decimal float is not supported for this target, ignored
configure: WARNING: fixed-point is not supported for this target, ignored
--with-abi=lp64d is not supported for ISA rv32ima
make[2]: *** [configure-gcc] Error 1
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib] Error 2
```

aswaterman, 2016-12-13:
This is fixed on the branch I'm about to merge into master https://github.com/riscv/riscv-tools/tree/bump-gcc

aopenna, 2016-12-22:
I have the same problem when building. @aswaterman  did you need any help on your current work on this issue?

aswaterman, 2016-12-23:
I will fix this tonight

aopenna, 2016-12-23:
Thanks a lot!!

aswaterman, 2016-12-23:
So, I can't merge this upstream tonight, because of an unrelated problem with riscv-tools.  In the mean time, check out this branch https://github.com/riscv/riscv-tools/tree/bump-gcc -- build.sh will fail, but you should have better luck with build-rv32ima.sh in the mean time.

aswaterman, 2016-12-23:
By the way, Merry Christmas.

aopenna, 2016-12-23:
Merry Christimas to you too! thanks a lot, i willl try it.

aopenna, 2016-12-23:
Hi @aswaterman. I had some small errors compiling risc-isa-sim, errors on gdbserver.cc and debugmodule.cc. Attached are the files with my corrections. If you think it is relevant fix
[riscv-tools.tar.gz](https://github.com/riscv/riscv-tools/files/670980/riscv-tools.tar.gz)
, please commit it to bump-gcc branch since I a noob on git and don't know how to commit changes on submodules.

timsifive, 2017-12-20:
This works for me today.

closed 2017-12-20 by timsifive