riscv-software-src/riscv-tools #53: riscv-tests wont make when toolchain is rv32ima

aopenna, 2016-12-27:
Hello team,
I build the riscv-tools from branch bump-gcc in order to build rv32ima variant that i'm working on. Then I tried to build the riscv-tests from instructions, but a error raised during make:

`` mm.o: na função `.L5':
mm.c:(.text+0x12c): referência indefinida para `fma'
mm.c:(.text+0x154): referência indefinida para `fma'
mm.c:(.text+0x184): referência indefinida para `fma'
mm.c:(.text+0x1b8): referência indefinida para `fma'
mm.o: na função `.L7':
mm.c:(.text+0x260): referência indefinida para `fma'
mm.o:mm.c:(.text+0xd4c): mais referências indefinidas para seguir `fma'
collect2: error: ld returned 1 exit status
/home/aopenna/RISCV/riscv-tools/riscv-tests/benchmarks/mm/bmark.mk:28: recipe for target 'mm.riscv' failed
make[1]: *** [mm.riscv] Error 1
make[1]: Leaving directory '/home/aopenna/RISCV/riscv-tools/riscv-tests/benchmarks'
Makefile:18: recipe for target 'benchmarks' failed
make: *** [benchmarks] Error 2``

agent001, 2017-06-27:
Build error in riscv-tests

Configuring project riscv-tests
configure: WARNING: using cross tools not prefixed with host triplet
Building project riscv-tests
/rocket-chip/riscv-tools/lib/gcc/riscv64-unknown-elf/7.1.1/../../../../riscv64-unknown-elf/bin/ld:/rocket-chip/riscv-tools/riscv-tests/build/../isa/../env/v/link.ld:0: syntax error
collect2: error: ld returned 1 exit status
make[1]: *** [rv32ui-v-simple] Error 1
make: *** [isa] Error 2
make: *** Waiting for unfinished jobs....

jim-wilson, 2018-02-03:
There are two different problems reported here.  I am not able to reproduce either.

For the first one, from  aopenna, fma should come from the math library.  The makefile is providing -lm when linking, so this should work.  But maybe it was broken earlier.  You can find more info about what happened by looking in the riscv-tests/build/build.log file.

For the second one, from agent001, env/v/link.ld looks like a valid linker script file, but maybe there was a bug earlier.  Likewise, you can find more info about what happened by looking in the riscv-tests/builds/build.log file.

closed 2018-02-03 by jim-wilson