riscv-software-src/riscv-tools #54: building master on macOS X

michaeljclark, 2017-01-05:
I've successfully built master on Linux however I am having trouble building master on macOS X.

Here is my stamps dir:

```
ls stamps/
build-binutils-newlib
```

It's failing on the second step: `build-gcc-newlib`

```
mclark@monty:/Volumes/riscv-toolchain-build-master/riscv-tools/riscv-gnu-toolchain/build$ make
rm -rf stamps/build-gcc-newlib build-gcc-newlib
mkdir build-gcc-newlib
cd build-gcc-newlib && /Volumes/riscv-toolchain-build-master/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/configure \
		--target=riscv64-unknown-elf \
		--prefix=/opt/riscv/toolchain-master \
		--disable-shared \
		--disable-threads \
		--enable-tls \
		--enable-languages=c,c++ \
		--with-newlib \
		--disable-libmudflap \
		--disable-libssp \
		--disable-libquadmath \
		--disable-libgomp \
		--disable-nls \
		--disable-soft-float \
		--disable-multilib \
		
/bin/sh: /Volumes/riscv-toolchain-build-master/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/configure: No such file or directory
make: *** [stamps/build-gcc-newlib] Error 127
```

I noticed that the subdirectory exists. It's almost as if one of the copy commands is working differently on macOS X has there is an additional subdirectory `newlib-gcc/newlib` containing the source instead of `newlib-gcc` which is what the Makefile is expecting.

```
mclark@monty:/Volumes/riscv-toolchain-build-master/riscv-tools/riscv-gnu-toolchain/build$ ls /Volumes/riscv-toolchain-build-master/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/configure
/Volumes/riscv-toolchain-build-master/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/newlib/configure
```

Any ideas on how to troubleshoot?

I really only need binutils for the latest CSRs so I could perhaps overlay the binutils on top of an older gcc...

michaeljclark, 2017-01-05:
OK. I haven't solved the build issue but I am okay for now. I made a copy of my earlier toolchain and copied over the binutils that successfully built from master, so now I have access to the latest CSRs. **on macOS X**. gcc seems happy with the new CSRs. I have newer gas. Last build is older than I thought. I called the combined toolchain frank 😄 .

```
$ ls -ltr /opt/riscv/toolchain-frank/bin/
total 105232
-rwxr-xr-x  1 mclark  admin   882868 Jan 11  2016 riscv64-unknown-elf-gcc-5.2.0
-r-xr-xr-x  1 mclark  admin     9964 Jul  3  2016 fesvr-zedboard
-r-xr-xr-x  1 mclark  admin     9948 Jul  3  2016 fesvr-rs232
-r-xr-xr-x  1 mclark  admin    10072 Jul  3  2016 fesvr-eth
-rwxr-xr-x  1 mclark  admin    39568 Jul  3  2016 riscv64-unknown-elf-gcc-ranlib
-rwxr-xr-x  1 mclark  admin    39560 Jul  3  2016 riscv64-unknown-elf-gcc-nm
-rwxr-xr-x  1 mclark  admin    39472 Jul  3  2016 riscv64-unknown-elf-gcc-ar
-rwxr-xr-x  1 mclark  admin   960440 Jul  3  2016 riscv64-unknown-elf-gcc-6.1.0
-rwxr-xr-x  1 mclark  admin   960440 Jul  3  2016 riscv64-unknown-elf-gcc
-rwxr-xr-x  1 mclark  admin   960448 Jul  3  2016 riscv64-unknown-elf-g++
-rwxr-xr-x  1 mclark  admin   960456 Jul  3  2016 riscv64-unknown-elf-cpp
-rwxr-xr-x  1 mclark  admin   960448 Jul  3  2016 riscv64-unknown-elf-c++
-rwxr-xr-x  1 mclark  admin   511972 Jul  3  2016 riscv64-unknown-elf-gcov-tool
-rwxr-xr-x  1 mclark  admin   563972 Jul  3  2016 riscv64-unknown-elf-gcov
-rwxr-xr-x  1 mclark  admin  1163556 Aug  6 16:58 riscv32-unknown-elf-strip
-rwxr-xr-x  1 mclark  admin   921192 Aug  6 16:58 riscv32-unknown-elf-strings
-rwxr-xr-x  1 mclark  admin   926284 Aug  6 16:58 riscv32-unknown-elf-size
-rwxr-xr-x  1 mclark  admin   569532 Aug  6 16:58 riscv32-unknown-elf-readelf
-rwxr-xr-x  1 mclark  admin   963256 Aug  6 16:58 riscv32-unknown-elf-ranlib
-rwxr-xr-x  1 mclark  admin  1350408 Aug  6 16:58 riscv32-unknown-elf-objdump
-rwxr-xr-x  1 mclark  admin  1163564 Aug  6 16:58 riscv32-unknown-elf-objcopy
-rwxr-xr-x  1 mclark  admin   933996 Aug  6 16:58 riscv32-unknown-elf-nm
-rwxr-xr-x  1 mclark  admin    78564 Aug  6 16:58 riscv32-unknown-elf-elfedit
-rwxr-xr-x  1 mclark  admin   918988 Aug  6 16:58 riscv32-unknown-elf-c++filt
-rwxr-xr-x  1 mclark  admin   963232 Aug  6 16:58 riscv32-unknown-elf-ar
-rwxr-xr-x  1 mclark  admin   921112 Aug  6 16:58 riscv32-unknown-elf-addr2line
-rwxr-xr-x  1 mclark  admin  1292948 Aug  6 16:58 riscv32-unknown-elf-as
-rwxr-xr-x  1 mclark  admin   997516 Aug  6 16:58 riscv32-unknown-elf-gprof
-rwxr-xr-x  1 mclark  admin  1447580 Aug  6 16:58 riscv32-unknown-elf-ld.bfd
-rwxr-xr-x  1 mclark  admin  1447580 Aug  6 16:58 riscv32-unknown-elf-ld
-rwxr-xr-x  1 mclark  admin   961104 Aug  6 17:17 riscv32-unknown-elf-g++
-rwxr-xr-x  1 mclark  admin   961104 Aug  6 17:17 riscv32-unknown-elf-c++
-rwxr-xr-x  1 mclark  admin   512444 Aug  6 17:17 riscv32-unknown-elf-gcov-tool
-rwxr-xr-x  1 mclark  admin   564428 Aug  6 17:17 riscv32-unknown-elf-gcov
-rwxr-xr-x  1 mclark  admin    39680 Aug  6 17:17 riscv32-unknown-elf-gcc-ranlib
-rwxr-xr-x  1 mclark  admin    39672 Aug  6 17:17 riscv32-unknown-elf-gcc-nm
-rwxr-xr-x  1 mclark  admin    39576 Aug  6 17:17 riscv32-unknown-elf-gcc-ar
-rwxr-xr-x  1 mclark  admin   961088 Aug  6 17:17 riscv32-unknown-elf-gcc-6.1.0
-rwxr-xr-x  1 mclark  admin   961088 Aug  6 17:17 riscv32-unknown-elf-gcc
-rwxr-xr-x  1 mclark  admin   961112 Aug  6 17:17 riscv32-unknown-elf-cpp
-r-xr-xr-x  1 mclark  admin    11856 Jan  6 05:03 elf2hex
-r-xr-xr-x  1 mclark  admin     9744 Jan  6 05:05 xspike
-r-xr-xr-x  1 mclark  admin     9228 Jan  6 05:05 termios-xspike
-r-xr-xr-x  1 mclark  admin    26540 Jan  6 05:05 spike-dasm
-r-xr-xr-x  1 mclark  admin    75436 Jan  6 05:05 spike
-rwxr-xr-x  1 mclark  admin  1223356 Jan  6 05:12 riscv64-unknown-elf-strip
-rwxr-xr-x  1 mclark  admin   976864 Jan  6 05:12 riscv64-unknown-elf-strings
-rwxr-xr-x  1 mclark  admin   977860 Jan  6 05:12 riscv64-unknown-elf-size
-rwxr-xr-x  1 mclark  admin   598420 Jan  6 05:12 riscv64-unknown-elf-readelf
-rwxr-xr-x  1 mclark  admin  1018952 Jan  6 05:12 riscv64-unknown-elf-ranlib
-rwxr-xr-x  1 mclark  admin  1414416 Jan  6 05:12 riscv64-unknown-elf-objdump
-rwxr-xr-x  1 mclark  admin  1223356 Jan  6 05:12 riscv64-unknown-elf-objcopy
-rwxr-xr-x  1 mclark  admin   989676 Jan  6 05:12 riscv64-unknown-elf-nm
-rwxr-xr-x  1 mclark  admin    82700 Jan  6 05:12 riscv64-unknown-elf-elfedit
-rwxr-xr-x  1 mclark  admin   970564 Jan  6 05:12 riscv64-unknown-elf-c++filt
-rwxr-xr-x  1 mclark  admin  1018928 Jan  6 05:12 riscv64-unknown-elf-ar
-rwxr-xr-x  1 mclark  admin   976784 Jan  6 05:12 riscv64-unknown-elf-addr2line
-rwxr-xr-x  1 mclark  admin  1352748 Jan  6 05:12 riscv64-unknown-elf-as
-rwxr-xr-x  1 mclark  admin  1053340 Jan  6 05:12 riscv64-unknown-elf-gprof
-rwxr-xr-x  1 mclark  admin  1507332 Jan  6 05:12 riscv64-unknown-elf-ld.bfd
-rwxr-xr-x  1 mclark  admin  1507332 Jan  6 05:12 riscv64-unknown-elf-ld
-rwxr-xr-x  1 mclark  admin  1211684 Jan  6 05:12 riscv64-unknown-elf-run
-rwxr-xr-x  1 mclark  admin  6540016 Jan  6 05:12 riscv64-unknown-elf-gdb
```

michaeljclark, 2017-01-05:
BTW just a note - I first tried using hex constants for the newer CSRs but it didn't work. In any case, it is working for me now.

```
#define MIDELEG 0x303
#define MEDELEG 0x302
#define HIDELEG 0x203
#define HEDELEG 0x202

// send S-mode interrupts and most exceptions straight to S-mode
static void delegate_traps()
{
  uintptr_t interrupts = MIP_SSIP | MIP_STIP | MIP_SEIP;
  uintptr_t exceptions =
    (1U << CAUSE_MISALIGNED_FETCH) |
    (1U << CAUSE_FAULT_FETCH) |
    (1U << CAUSE_BREAKPOINT) |
    (1U << CAUSE_FAULT_LOAD) |
    (1U << CAUSE_FAULT_STORE) |
    (1U << CAUSE_BREAKPOINT) |
    (1U << CAUSE_USER_ECALL);

  write_csr(MIDELEG, interrupts);
  write_csr(HIDELEG, interrupts);
  write_csr(MEDELEG, exceptions);
  write_csr(HEDELEG, exceptions);
  assert(read_csr(MIDELEG) == interrupts);
  assert(read_csr(HIDELEG) == interrupts);
  assert(read_csr(MEDELEG) == exceptions);
  assert(read_csr(HEDELEG) == exceptions);
}
```

```
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks  -DBBL_PAYLOAD=\"/Users/mclark/src/riscv-tools/riscv-meta/linux-4.6.2/vmlinux\" -I. -I../bbl -c ../bbl/minit.c
../bbl/minit.c: Assembler messages:
../bbl/minit.c:55: Error: Instruction csrw requires absolute expression
../bbl/minit.c:56: Error: Instruction csrw requires absolute expression
../bbl/minit.c:57: Error: Instruction csrw requires absolute expression
../bbl/minit.c:58: Error: Instruction csrw requires absolute expression
../bbl/minit.c:59: Error: Instruction csrr requires absolute expression
../bbl/minit.c:60: Error: Instruction csrr requires absolute expression
../bbl/minit.c:61: Error: Instruction csrr requires absolute expression
../bbl/minit.c:62: Error: Instruction csrr requires absolute expression
make: *** [minit.o] Error 1
```

michaeljclark, 2017-01-11:
It may be that a submodule is out of sync after a pull e.g. riscv-gnu-toolchain/riscv-newlib

I'll try building a clean checkout of riscv-tools:master on macos and will update this issue...

palmer-dabbelt, 2017-08-26:
We just built on OSX, and since you didn't update I'm assuming it's user error :).  Feel free to re-open it I screwed up.

closed 2017-08-26 by palmer-dabbelt

michaeljclark, 2017-08-26:
No worries. I'll try a build on macOS with a clean checkout. I currently still have gcc-6.1.0 on my mac :) I've found vintage compilers are useful for comparisons...

```
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

michaeljclark, 2017-08-26:
It's working fine now. I built from the `v20170818` tag. Thanks.

```
$ uname -a
Darwin monty.local 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 7.1.1 20170509
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```