riscv-software-src/riscv-tools #55: Build script throws error

LyronDev, 2017-02-03:
I build the complete toolchain before. Last year in December.

I followed the exact same steps, but this time ran into a build error when executing:
$ ./build.sh

`/home/usr/RISCVTools/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/lto-compress.c:34:18: fatal error: zlib.h: No such file or directory
 #include <zlib.h>
                  ^
compilation terminated.
make[3]: *** [lto-compress.o] Error 1
make[2]: *** [all-gcc] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib] Error 2
`

I look in the folder for the missing header zlib.h and it's nowhere to find.
Did i miss something?

aswaterman, 2017-02-03:
Did you install the zlib1g-dev package (or equivalent for your platform) as README.md says?

LyronDev, 2017-02-03:
Thanks for the "quicker than lightning" response. That was indeed the package i was missing. 
Just noticed that this has been added to the requirements.
Thank you very much!!

closed 2017-02-03 by LyronDev