riscv-software-src/riscv-tools #56: Build fails, Dell XPS13, Ubuntu 16.04, on newlib

AlexRast, 2017-02-09:
Normal following-the-directions style build - define RISCV, export RISCV in path and as variable, install all Ubuntu Linux dependencies, run build.sh, produces the following:

make[3]: *** No rule to make target '../build-x86_64-unknown-linux-gnu/libcpp/libcpp.a', needed by 'build/genmatch'. Stop.
make[2]: *** [all-gcc] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib] Error 2

Full gory details of output below.
But it seems either the build is expecting libcpp.a was already built or something is wrong in the makefile.

My experience suggests that probably this is a case of an expected environment variable that doesn't exist on my system. Or an expected script that wasn't run. 

I *hope* it isn't a situation where the build is designed so that you mustn't set the RISCV directory to the same path as the location of the build.sh file (i.e. you must build into an entirely different directory from the source) 

Full output from build below
----------------------------------------------------------------------------------------------------------------------------------------------
Starting RISC-V Toolchain build process

Removing existing riscv-fesvr/build directory
Configuring project riscv-fesvr
Building project riscv-fesvr
Installing project riscv-fesvr
mkdir /home/alex_rast/Local_prg/riscv/include
mkdir /home/alex_rast/Local_prg/riscv/include/fesvr
mkdir /home/alex_rast/Local_prg/riscv/lib
mkdir /home/alex_rast/Local_prg/riscv/bin
mkdir /home/alex_rast/Local_prg/riscv/lib/pkgconfig

Configuring project riscv-isa-sim
Building project riscv-isa-sim
Installing project riscv-isa-sim
mkdir /home/alex_rast/Local_prg/riscv/include/spike

Configuring project riscv-gnu-toolchain
Building project riscv-gnu-toolchain
ar: `u' modifier ignored since `D' is the default (see `U')
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/sysinfo.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
In file included from /home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/syslex_wrap.c:25:0:
syslex.c: In function ‘yy_get_next_buffer’:
syslex.c:1226:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  if ((int) ((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_s
                                            ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/defparse.y: warning: 27 shift/reduce conflicts [-Wconflicts-sr]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/rcparse.y: warning: 58 shift/reduce conflicts [-Wconflicts-sr]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/rcparse.y: warning: 10 reduce/reduce conflicts [-Wconflicts-rr]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/mcparse.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/doc/binutils.texi:788: warning: @ref node name should not contain `.'
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/doc/binutils.texi:1164: warning: @xref node name should not contain `.'
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/doc/binutils.texi:3678: warning: @ref node name should not contain `.'
arlex.c: In function ‘yy_get_next_buffer’:
arlex.c:1352:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  if ((int) ((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_s
                                            ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/ld/deffilep.y: warning: 76 shift/reduce conflicts [-Wconflicts-sr]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/ld/ld.texinfo:8145: warning: @pxref node name should not contain `.'
In file included from /home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/ld/ldlex-wrapper.c:26:0:
ldlex.c: In function ‘yy_get_next_buffer’:
ldlex.c:3497:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  if ((int) ((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_s
                                            ^
configure: WARNING: decimal float is not supported for this target, ignored
ar: `u' modifier ignored since `D' is the default (see `U')
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/complete.c: In function ‘fnwidth’:
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/complete.c:701:12: warning: implicit declaration of function ‘wcwidth’ [-Wimplicit-function-declaration]
        w = wcwidth (wc);
            ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/display.c: In function ‘rl_redisplay’:
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/../riscv-binutils-gdb/readline/display.c:768:15: warning: implicit declaration of function ‘wcwidth’ [-Wimplicit-function-declaration]
        temp = wcwidth (wc);
               ^
[... 104 lines cut ...]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-common.c:6109:18: warning: format not a string literal and no format arguments [-Wformat-security]
     error (gmsgid);
                  ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-common.c:6113:21: warning: format not a string literal and no format arguments [-Wformat-security]
       error (message);
                     ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/cilk.c: In function ‘bool check_no_cilk(tree, const char*, const char*, location_t)’:
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/cilk.c:1435:33: warning: format not a string literal and no format arguments [-Wformat-security]
       error_at (loc, array_msgid);
                                 ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/cilk.c:1441:33: warning: format not a string literal and no format arguments [-Wformat-security]
       error_at (loc, spawn_msgid);
                                 ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-warn.c: In function ‘void warn_for_restrict(unsigned int, vec<tree_node*, va_gc>*)’:
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-warn.c:2216:52: warning: unknown conversion type character ‘Z’ in format [-Wformat=]
     param_pos + 1, arg_positions, arg_positions_len);
                                                    ^
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-warn.c:2216:52: warning: too many arguments for format [-Wformat-extra-args]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-warn.c:2216:52: warning: unknown conversion type character ‘Z’ in format [-Wformat=]
/home/alex_rast/Local_prg/riscv/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/c-family/c-warn.c:2216:52: warning: too many arguments for format [-Wformat-extra-args]
make[3]: *** No rule to make target '../build-x86_64-unknown-linux-gnu/libcpp/libcpp.a', needed by 'build/genmatch'. Stop.
make[2]: *** [all-gcc] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib] Error 2
-------------------------------------------------------------------------------------------------------------------------------------------

sorear, 2017-02-09:
Build output will be easier to read in a

    ```
    verbatim block
    ```

aswaterman, 2017-02-09:
Doesn't seem like a RISC-V problem.  Have you successfully built an x86 toolchain on this machine?

AlexRast, 2017-02-14:
I need to understand some terminology here. When you say 'built an x86 toolchain' in this context, do you mean building a compiler targetting any platform, to operate on x86 (i.e. a cross-compiler such as e.g. one that compiles ARM-native applications on an x86 machine), or a native compiler to build applications for the machine also doing the compilation (x86 target on Ubuntu)? 

I've built several cross-compiler without problems. I also have a working toolchain for native x86 compilation in Ubuntu, although I believe this would not use the 'unknown' target. 

However I should note that my level of knowledge about the gcc build is not *nearly* at the level that would enable me to speculate on what the problem related to in any direct way. So regardless of where the problem may lie I am going to need more concrete direction if I am to have any chance of fixing it in meaningful time.

aswaterman, 2017-02-14:
I should've more generically asked "have you built any GCC on this machine".

To isolate RISC-V-specific stuff, I'd try building a native gcc with this exact version of gcc.  You can try something like ```mkdir test-native && cd test-native && ../riscv-gcc/configure --disable-mult
ilib --prefix=/tmp && make``` from within riscv-gnu-toolchain.  Whether that succeeds or fails will give us more information.

AlexRast, 2017-02-15:
Yes, the build succeeded without problems. Making a guess, I suspect this may be related to the fact that the makefile produced gives: host=x86_64-pc-linux-gnu and likewise for target. So it's not trying to build x86_64-unknown-linux-gnu targets. However this is speculation on my part.

jim-wilson, 2018-02-03:
An error like this can occur if you run configure or make in the riscv-gcc/gcc subdir instead of in the riscv-gcc dir.  However, this should not have happened if using the build scripts.

It isn't clear how to reproduce.  The provided info only contains the stderr from the build.  Most of the info is in the riscv-gnu-toolchain/build/build.log file which was not provided.

Given the length of time this has been open, and the fact that there have been so many source and script file changes since then, I don't think it serves any purpose to keep this open any longer.

closed 2018-02-03 by jim-wilson