riscv-software-src/riscv-tools #59: When weill riscv-tools support RV32EMC?

zhenbohu, 2017-03-17:
We are a company focus on some mixed-signal sensor product, we are very exciting about RISC-V and plan to customize design an in-house ultra low power RISCV core with some special custom instructions. In this deep embedded IoT application, RV32EMC is the best ISA choice. However when we build the riscv-tools, the error reported that the RV32EMC is not supported now. This is very depressing to us. We believe RISCV is going to be brilliantly in deep embedded IoT domain, hence RV32EMC is very important. So sincerely may I have a question when RV32EMC will be supported in riscv-tool? 

[image]

aswaterman, 2017-03-17:
We want to support RV32E, but at this moment, it is not a high-priority item for @palmer-dabbelt and me.  @kito-cheng are you guys planning on adding RV32E support to GCC?

kito-cheng, 2017-03-17:
We have initial support for gcc, binutils, newlib and gdb-sim for RV32E, but it's low priority task for us too, maybe April or May will contribute back.

zhenbohu, 2017-03-17:
Thanks for you guys quick response @aswaterman @kito-cheng 
As a fans of RISCV, I think IoT deep embedded domain is really a good opportunity for RISCV, the RV32E relevant support should be promoted with more higher priority and put into place as soon as possible. We will be very looking forward for this feature.

aswaterman, 2017-03-17:
@zhenbohu we agree!  The challenge is that *everything* seems to be high-priority now.

zhenbohu, 2017-03-17:
Thank you. Is there any earlier branch for this feature? @aswaterman 
Please let us know when there is, we can do some earlier testing base on that.

zhenbohu, 2017-04-26:
Hi, Dear Guys

May I ask is there any progress on RV32E supporting on tool chain?

Thanks
Bob

kito-cheng, 2017-06-16:
RV32E is review on progress now:
GCC: https://github.com/riscv/riscv-gcc/pull/77
Binutils: https://github.com/riscv/riscv-binutils-gdb/pull/87
newlib: https://github.com/riscv/riscv-newlib

palmer-dabbelt, 2017-06-16:
@kito-cheng Do you mind sending a message to sw-dev announcing this?  It would be good to get everyone's eyes on it -- particularly the LLVM people, as we're trying to ensure LLVM and GCC are as compatible as possible on RISC-V.

david-cureton-baker, 2017-10-28:
Any new status on rv32e, rv32ec status?

palmer-dabbelt, 2017-10-28:
We have some patches floating around.  If you want to play around with RV32E right now then you should be able to use what's on riscv-next, but be aware that we haven't finalized it yet so things like the ABI still might change a bit around the edges.  We also don't run RV32E targets as part of the regressions yet, so it won't be up the the stability level of the other targets.

In other words: most of the code is there, it should work in many cases, but don't ship binaries yet.

jm4rtin, 2017-11-09:
There is a pull request in riscv-binutils-gdb (https://github.com/riscv/riscv-binutils-gdb/pull/103/commits) that adds support for RV32E and it was marked merged into riscv-next; however, when looking at one of the files (gas/config/tc-riscv.c) the commit is not even listed in the history (https://github.com/riscv/riscv-binutils-gdb/commits/riscv-next/gas/config/tc-riscv.c). Nor does there seem to be a commit on that day for RV32E support: https://github.com/riscv/riscv-binutils-gdb/commits/riscv-next?after=0b05400ac71805ef5229c40b7674150027af0fbd+979

Has the history of riscv-next been rewritten? As it stands now there is no RV32E support in riscv-next.

palmer-dabbelt, 2017-11-09:
Oh, sorry about that -- we've been re-organizing all the branches and it must have just disappeared.  It should be back now.  Note that riscv-next is a staging branch, so it regularly gets rebased as patches become more stable.

jm4rtin, 2017-11-09:
I see what's going on now. Any clue what is missing in riscv-next for the GDB build?

/local/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/riscv-tdep.c: In function ‘void riscv_extract_return_value(type*, regcache*, gdb_byte*, int)’:
/local/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/riscv-tdep.c:336:52: error: ‘get_regcache_arch’ was not declared in this scope
   struct gdbarch *gdbarch = get_regcache_arch (regs);
                                                    ^
/local/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/riscv-tdep.c: In function ‘void riscv_store_return_value(type*, regcache*, const gdb_byte*, int)’:
/local/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/riscv-tdep.c:364:52: error: ‘get_regcache_arch’ was not declared in this scope
   struct gdbarch *gdbarch = get_regcache_arch (regs);
                                                    ^
gmake[3]: *** [riscv-tdep.o] Error 1
gmake[2]: *** [all-gdb] Error 2
gmake[1]: *** [all] Error 2
gmake: *** [stamps/build-binutils-newlib] Error 2

palmer-dabbelt, 2017-11-09:
Our GDB port isn't upstream yet, which means when other people refactor code it doesn't get applied to our port.  In general, you can figure out how to fix missing build things like this by just looking through the git log.  Here's a commit that looks suspicious

````
commit ac7936dfd0c85e5de2dfec45ca0dbf72baeffa51
Author: Yao Qi <yao.qi@linaro.org>
Date:   Wed Oct 25 16:37:03 2017 +0100

    s/get_regcache_arch (regcache)/regcache->arch ()/g
    
    This patches removes get_regcache_arch, and use regache->arch () instead.
    The motivation of this change is that I am going to move some basic stuff
    into a base class of regcache.  I don't need to update "client" code
    regcache->arch ().  On the other hand, this patch shortens the code a
    little bit.
````

We probably just need a fix in our port for that.