riscv-software-src/riscv-tools #64: how to compile program with RVC?

theotuck, 2017-04-15:
hello,everyone.

I read  a report for RVC(https://people.eecs.berkeley.edu/~krste/papers/waterman-ms.pdf ), then I'd like to use the RVC to do the similar tests. But it is ambiguous for me to how to do those experiments. I get several problems:

1) the gcc could not generate the RVC code or assemble code automatically. even with the option -march. I don't know if I did something wrong, because there is no guide book.

2) I could get the RV assemble code, then I add the RVC assemble code to replace the RV assemble code. the assemble file can generate the elf file, but when I got the instruction list with "spike -l pk", it always tells me there are errors:
           -exception illegal instruction, it is a div instruction.
           -exception trap instruction access fault 
  firstly I just thought it was my fault because I modified the assemble file manually. But when I try the RV code, it returns the same error.The compiling and simulating options as follow:
           - riscv64-unknown-elf-gcc -march=RV64IFDC 
           - spike --isa=RV64IFDC -l pk

3) In my virtual machine, it is strange about the supporting instruction sets. 
            - Compile the code without the F/D sets, there is a assembler message error " Architecture doesn't allow hardfloat ABI"
            -Compile the code with M set, there is another assembler message error " Unsupported ISA subset M"
            - the RV64IFDC could get through the compile but do not get the correct executable file.
    I really get lost……

4) when I dump the executable file with riscv64-unknown-elf-objdump -d, I could not find the RVC instrucions I added. I guess they are all decoded into RV instructions. Is that ture?

5) Are there some options or command lines to tell the gcc to get the RVC assemble file automatically instead of modifying the normal RV assemble file manually?  from the report I give above, it mentioned a RVC assembler, but I reeeeeeeeealy do not know how it work and, where the assembler is.

If I did something wrong, please tell me and help me to correct them.
So many thanks to any reply.

sorear, 2017-04-15:
(1) if you have a properly built gcc for Linux (I'm using the one created as a side effect of building sifive/freedom-u-sdk) RVC usage is automatic.  It can be disabled with `-march=rv64g` or force enabled with `-march=rv64gc`.

(2) the spike errors are because the isa needs to be rv64imafdc (lower case?  definitely need m for division) and you also need to pass a.out to spike.  I mostly use qemu though.

(3) if you're targeting an ISA without F/D you need to force soft-float.  `-march=rv64imafd` works for me though (must be lowercase)

(4) RVC instructions can be recognized in objdump output because they have 4 hex digits instead of 8.  Alternately pass `-Mno-aliases` to objdump to see the same syntax used in Andrew's report.

(5) if you pass `-march=rv64gc` the assembler will use RVC with no further action required.

theotuck, 2017-04-15:
So many thanks!

but I am afraid those options dose not work correctly as well.

when I add the `-march=RV64IMAFDC` (the gcc could not reconginize the RV64GC because ISA strings must begin with I), it could generate a executable file. But when I use `spike pk myProgram`, the pk/elf.c always gives a assert.

I know this is because the macro definetion,  __riscv_compressed, dose not be defined so that the pk can not get through the check.

But soon I find that if I run spike with the command line ` spike --isa=RV64IFDC pk myProgram`, the assertion disappears while the code jumps into a endless loop because the illegal instruction and trap_instruction_access_fault.

I check the illegal instruction, it is div, the division. I guess that lies in the lack of M set.  

Unfortunately, if I add the M set in the runtime, the pk will give me the assesstion.  Then I jump into a endless loop, too.

sorear, 2017-04-15:
As I said a could times, `-march=RV64IMAFDC` is wrong, `-march=rv64imafdc` is the only correct way.

theotuck, 2017-04-16:
Really appreciate your answers!

Following as you said, I find something different with my source code. 

when I try to use `-march=rv64imafdc', there is an error from my source file riscv-common.c. But the newest [one](https://github.com/riscv/riscv-gcc/blob/riscv-next/gcc/common/config/riscv/riscv-common.c) doesn't have.

And when it comes to the spike and pk, the problem still exist. I'll check them first.

Thank you again!

palmer-dabbelt, 2017-04-17:
This works for me

````
$ cat main.c 
#include <stdio.h>
int main() { printf("Hello, World!\n"); return 0; }
$ riscv64-unknown-elf-gcc main.c -o main -march=rv64imafdc -mabi=lp64d
$ riscv64-unknown-elf-objdump -d main | grep ld | head -n5
   10170:       60a2                    ld      ra,8(sp)
   10172:       6402                    ld      s0,0(sp)
   10198:       60a2                    ld      ra,8(sp)
   101ba:       60a2                    ld      ra,8(sp)
   101bc:       6402                    ld      s0,0(sp)
$ spike pk main
Hello, World!
````

I'm using the "priv-1.10" branch of riscv-tools, though this is expected to work everywhere (that's just what I had built).

theotuck, 2017-04-19:
Oh, yes. 
After updating and recompiling the source codes, my spike and programs have worked well now!
It's so nice for your help and reply!

closed 2017-04-19 by theotuck