riscv-software-src/riscv-tools #66: error when building riscv-gnu-toolchain

xinjic, 2017-04-21:
configure: WARNING: decimal float is not supported for this target, ignored
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/sim/riscv/configure: line 13590: SIM_AC_OPTION_HOSTENDIAN: command not found
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c:100:31: error: 'DEFAULT_GENERATE_ELF_STT_COMMON' undeclared here (not in a function)
 int flag_use_elf_stt_common = DEFAULT_GENERATE_ELF_STT_COMMON;
                               ^
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c: In function 'print_version_id':
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c:229:14: error: 'TARGET_ALIAS' undeclared (first use in this function)
     VERSION, TARGET_ALIAS, BFD_VERSION_STRING);
              ^
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c:229:14: note: each undeclared identifier is reported only once for each function it appears in
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c: In function 'parse_args':
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c:674:5: error: 'TARGET_ALIAS' undeclared (first use in this function)
     TARGET_ALIAS);
     ^
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c:689:44: error: 'TARGET_CANONICAL' undeclared (first use in this function)
    fprintf (stderr, _("canonical = %s\n"), TARGET_CANONICAL);
                                            ^
/home/jinxin/Projects/riscv/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gas/as.c:690:43: error: 'TARGET_CPU' undeclared (first use in this function)
    fprintf (stderr, _("cpu-type = %s\n"), TARGET_CPU);
                                           ^
make[5]: *** [as.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all-gas] Error 2
make[2]: *** Waiting for unfinished jobs....
configure: WARNING: no enhanced curses library found; disabling TUI
configure: WARNING: expat is missing or unusable; some features may be unavailable.
configure: WARNING: python is missing or unusable; some features may be unavailable.
configure: WARNING: linux/perf_event.h missing or too old; some features may be unavailable.
configure: WARNING: libipt is missing or unusable; some features may be unavailable.
configure: WARNING: babeltrace is missing or unusable; GDB is unable to read CTF data.
make[1]: *** [all] Error 2
make: *** [stamps/build-binutils-newlib] Error 2

so,how to deal with it? thank you sooooo much!

jimmysitu, 2017-07-02:
I have the same issue with Ubuntu 14.04. But I can make it pass when I use a 16.04.
So I think it is a issue with compiler version

labsin, 2017-07-23:
I had the same issue with the LibreElec toolchain.
Found the issue was that I had C_INCLUDE_PATH set system-wide in my .bashrc. It accidentally started with a ":" which I think was the problem.

jim-wilson, 2018-02-03:
I was able to reproduce the error by setting C_INCLUDE_PATH incorrectly.  This causes gas to include the wrong config.h file, of which there are multiple ones in the tree.

closed 2018-02-03 by jim-wilson

dinggaoliu, 2019-10-25:
> I had the same issue with the LibreElec toolchain.
> Found the issue was that I had C_INCLUDE_PATH set system-wide in my .bashrc. It accidentally started with a ":" which I think was the problem.

Thank you very much! I print "unset C_INCLUDE_PATH" and then my error was solved!