riscv-software-src/riscv-tools #70: No root device when trying running linux over Spike

MaMazav, 2017-05-03:
I followed the instructions in risc-v repositories in order to have linux running over Spike.
When trying to boot linux with the following command:
    spike +disk=path/to/linux-4.6.2/root.bin path/to/riscv-pk/build/bbl vmlinux
I get "Kernel panic". First interesting error in the output is:
    VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6

After some research I found some posts which claim that the block device is not supported anymore and I shouldn't use the +disk argument. Instead I should use initramfs. E.g.:
    http://stackoverflow.com/questions/37849574/how-to-rebuild-bbl-with-payload-option

I would be greatful to have instructions of how to use initramfs in order to make linux running over Spike (I'm not a linux-kernel expert).
Also I suggest to update the README.

Thanks

Full output follows:

              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] Linux version 4.6.2-00048-g9079be6 (idani@mtlxvdi032.yok.mtl.com) (gcc version 6.1.0 (GCC) ) #1 Wed Apr 26 13:54:38 IDT 2017
[    0.000000] Available physical memory: 2040MB
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080400000-0x00000000ffbfffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080400000-0x00000000ffbfffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080400000-0x00000000ffbfffff]
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 515100
[    0.000000] Kernel command line: 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2054268K/2088960K available (2104K kernel code, 110K rwdata, 428K rodata, 68K init, 218K bss, 34692K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=100000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] devtmpfs: initialized
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] NET: Registered protocol family 16
[    0.000000] vgaarb: loaded
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.000000] NET: Registered protocol family 2
[    0.000000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] TCP bind hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] TCP: Hash tables configured (established 16384 bind 16384)
[    0.000000] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.000000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.000000] NET: Registered protocol family 1
[    0.020000] console [sbi_console0] enabled
[    0.020000] futex hash table entries: 256 (order: 0, 6144 bytes)
[    0.020000] workingset: timestamp_bits=61 max_order=19 bucket_order=0
[    0.030000] jitterentropy: Initialization failed with host not compliant with requirements: 2
[    0.030000] io scheduler noop registered
[    0.030000] io scheduler cfq registered (default)
[    0.040000] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    0.040000] Please append a correct "root=" boot option; here are the available partitions:
[    0.040000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    0.040000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.6.2-00048-g9079be6 #1
[    0.040000] Call Trace:
[    0.040000] [<ffffffff80012fc8>] walk_stackframe+0x0/0xc8
[    0.040000] [<ffffffff80056e90>] panic+0xec/0x20c
[    0.040000] [<ffffffff800011d8>] mount_block_root+0x248/0x328
[    0.040000] [<ffffffff8000292c>] ksysfs_init+0x10/0x40
[    0.040000] [<ffffffff80001488>] prepare_namespace+0x148/0x198
[    0.040000] [<ffffffff80000da0>] kernel_init_freeable+0x1c8/0x200
[    0.040000] [<ffffffff8021b160>] rest_init+0x80/0x84
[    0.040000] [<ffffffff8021b174>] kernel_init+0x10/0x11c
[    0.040000] [<ffffffff8021b160>] rest_init+0x80/0x84
[    0.040000] [<ffffffff80011c0c>] ret_from_syscall+0x10/0x14
[    0.040000] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

palmer-dabbelt, 2017-05-03:
This is the expected behavior: there's no block devices in the system (because Spike doesn't emulate one), so you can't mount the root filesystem.  I usually build with an initramfs that just has busybox in it.  Here's some of the kernel documentation on how to build an initramfs https://www.kernel.org/doc/Documentation/filesystems/ramfs-rootfs-initramfs.txt .

MaMazav, 2017-05-04:
Tried with no success.
I used the content of the root.bin created by the instruction in the README of the riscv-linux repository. I entered the mounted directory and used:

    find . | cpio -o -H newc | gzip > ../initramfs.cpio.gz

In "make menuconfig" (again, in riscv-linux repository) I changed the initrmfs source file to "initramfs.cpio.gz" and ran "make -j4 ARCH=riscv vmlinux" again. Then ran spike but got the same output.
What is missing?

btw I see that the kernel command line is empty and has no "root=" argument:

    [ 0.000000] Built 1 zonelists in Zone order, mobility grouping on. Total pages: 515100
    [ 0.000000] Kernel command line:
    [ 0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)

Is it OK?

palmer-dabbelt, 2017-05-04:
This works whenever I do it.  I always pass a config file to the Linux build rather than my own CPIO archive, as I find that easier than remembering how to run cpio.

Here's my (horribly outdated) scripts: https://github.com/palmer-dabbelt/riscv-gentoo-infra

Did you re-build bbl?  The "--with-payload" option causes the given image to be built into bbl at build-time, so you'll need to rebuild it every time you change vmlinux.

MaMazav, 2017-05-04:
I did rebuild bbl.

I'll try your config file, thanks.

MaMazav, 2017-05-08:
Finally I got it work with some help of @noamc .
Start from creating a folder and fill it with the content of the ram drive:

    mkdir root
    cd root
    mkdir -p bin etc dev lib proc sbin sys tmp usr usr/bin usr/lib usr/sbin
    cp <busy-box-path>/busybox bin/busybox
    ln -s ../bin/busybox sbin/init
    ln -s sbin/init init
    sudo mknod dev/console c 5 1

In this folder, create a file etc/inittab with a content similar to the one shown in the riscv README:

    ::sysinit:/bin/busybox mount -t proc proc /proc
    ::sysinit:/bin/busybox mount -t tmpfs tmpfs /tmp
    ::sysinit:/bin/busybox mount -o remount,rw /dev/htifblk0 /
    ::sysinit:/bin/busybox --install -s
    #::respawn:/bin/busybox getty 38400 ttySBI0
    /dev/console::sysinit:-/bin/ash

Then create a cpio archive from this folder (run this command directly from the above created directory root):

    find . | cpio --quiet -o -H newc > <linux-repo>/rootfs.cpio

Now configure linux to embed the cpio file within the vmlinux. One way to do that is to type from the <linux-repo> directory:

    make ARCH=riscv menuconfig

Enter to General Setup, mark "Initial RAM filesystem and RAM disk". Then go to the option "Initramfs source file" and press enter to change it to "rootfs.cpio". Then Exit all the way back and save to .config.

Don't forget to rebuild vmlinux by "make -j4 ARCH=riscv vmlinux" and clean and rebuild bbl before rerunning Spike.

closed 2017-05-08 by MaMazav