riscv-software-src/riscv-tools #72: Support --with-{gmp,mpfr,mpc}

bschommer, 2017-05-10:
It would be nice if it was possible to pass these options to the configure scripts since allow it to use a local installation of newer versions of gmp,mpc and mpfr without root rights.

palmer-dabbelt, 2017-05-10:
Is this just for GCC?  If so, you're not supposed to do that, but are
instead supposed to call contrib/download_prerequisites.  We have an
autoconf widget that's supposed to be doing that in riscv-gnu-toolchain, is
it not detecting your system doesn't have the required libraries?

bschommer, 2017-05-10:
No it does not work it currently fails with:

configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
ftp://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header

palmer-dabbelt, 2017-05-10:
Can you post your config.log from riscv-gnu-toolchain/build?  I'll look at
it when I get to a computer.

For now you can just run download_prerequisites manually.

bschommer, 2017-05-10:
With calling the download_prequisites manually it worked.

palmer-dabbelt, 2017-05-10:
Can I have riscv-gnu-toolchain/build/config.log and riscv-gnu-toolchain/build/build-gcc-newlib-stage1/config.log?  That way I'll be able to see what autoconf did on your system, as this all builds for us.

bschommer, 2017-05-10:
riscv-gnu-toolchain/build/config.log:

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ../configure --prefix=/local/schommer/trunk/build/riscv-gcc.default/usr

## --------- ##
## Platform. ##
## --------- ##

hostname = zoidberg
uname -m = x86_64
uname -r = 4.4.46-11-default
uname -s = Linux
uname -v = #1 SMP Sat Feb 4 06:07:41 UTC 2017 (f555e41)

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/bin
PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/bin
PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/share/texlive/bin/x86_64-linux
PATH: /opt/rh/devtoolset-3/root/usr/bin
PATH: /opt/rh/python27/root/usr/bin
PATH: /opt/rh/git19/root/usr/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1876: checking for gcc
configure:1892: found /opt/rh/devtoolset-3/root/usr/bin/gcc
configure:1903: result: gcc
configure:2132: checking for C compiler version
configure:2141: gcc --version >&5
gcc (GCC) 4.9.2 20150212 (Red Hat 4.9.2-6)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2152: $? = 0
configure:2141: gcc -v >&5
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/rh/devtoolset-3/root/usr/libexec/gcc/x86_64-redhat-linux/4.9.2/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/opt/rh/devtoolset-3/root/usr --mandir=/opt/rh/devtoolset-3/root/usr/share/man --infodir=/opt/rh/devtoolset-3/root/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --enable-languages=c,c++,fortran,lto --enable-plugin --with-linker-hash-style=gnu --enable-initfini-array --disable-libgcj --with-isl=/builddir/build/BUILD/gcc-4.9.2-20150212/obj-x86_64-redhat-linux/isl-install --with-cloog=/builddir/build/BUILD/gcc-4.9.2-20150212/obj-x86_64-redhat-linux/cloog-install --with-mpc=/builddir/build/BUILD/gcc-4.9.2-20150212/obj-x86_64-redhat-linux/mpc-install --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.9.2 20150212 (Red Hat 4.9.2-6) (GCC) 
configure:2152: $? = 0
configure:2141: gcc -V >&5
[... 262 lines cut ...]
multilib_names=''
oldincludedir='/usr/include'
pdfdir='${docdir}'
prefix='/local/schommer/trunk/build/riscv-gcc.default/usr'
program_transform_name='s,x,x,'
psdir='${docdir}'
runstatedir='${localstatedir}/run'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""

configure: exit 0

bschommer, 2017-05-10:
riscv-gnu-toolchain/build/build-gcc-newlib/config.log:

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by configure, which was
generated by GNU Autoconf 2.64.  Invocation command line was

  $ /local/schommer/trunk/build/riscv-gcc.default/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/configure --target=riscv64-unknown-elf --prefix=/local/schommer/trunk/build/riscv-gcc.default/usr --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64g

## --------- ##
## Platform. ##
## --------- ##

hostname = zoidberg
uname -m = x86_64
uname -r = 4.4.46-11-default
uname -s = Linux
uname -v = #1 SMP Sat Feb 4 06:07:41 UTC 2017 (f555e41)

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/bin
PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/bin
PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/bin
PATH: /local/schommer/trunk/build/riscv-gcc.default/usr/share/texlive/bin/x86_64-linux
PATH: /opt/rh/devtoolset-3/root/usr/bin
PATH: /opt/rh/python27/root/usr/bin
PATH: /opt/rh/git19/root/usr/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2297: checking build system type
configure:2311: result: x86_64-pc-linux-gnu
configure:2358: checking host system type
configure:2371: result: x86_64-pc-linux-gnu
configure:2391: checking target system type
configure:2404: result: riscv64-unknown-elf
configure:2458: checking for a BSD-compatible install
configure:2526: result: /usr/bin/install -c
configure:2537: checking whether ln works
configure:2559: result: yes
configure:2563: checking whether ln -s works
configure:2567: result: yes
configure:2574: checking for a sed that does not truncate output
configure:2638: result: /bin/sed
configure:2647: checking for gawk
configure:2674: result: /usr/bin/gawk
configure:3210: checking for libatomic support
configure:3216: result: no
configure:3229: checking for libcilkrts support
configure:3235: result: no
[... 644 lines cut ...]

## ------------------- ##
## File substitutions. ##
## ------------------- ##

alphaieee_frag='/dev/null'
host_makefile_frag='/dev/null'
ospace_frag='/dev/null'
serialization_dependencies='serdep.tmp'
target_makefile_frag='/dev/null'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME ""
#define PACKAGE_TARNAME ""
#define PACKAGE_VERSION ""
#define PACKAGE_STRING ""
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define LT_OBJDIR ".libs/"

configure: exit 0

bschommer, 2017-05-10:
Both configs are for the build run with calling download_prequisites manually.

palmer-dabbelt, 2017-05-10:
OK, sorry, I think I just screwed it up.  Thanks for the bug report!

I've opened a riscv-gnu-toolchain pull request: https://github.com/riscv/riscv-gnu-toolchain/pull/240 .  Since this is a riscv-gnu-toolchain issue I'm going to close this bug, if that doesn't work then please post over there.

Thanks!

closed 2017-05-10 by palmer-dabbelt

advancedwebdeveloper, 2020-11-17:
I am looking forward to see anything acceptable for https://go.googlesource.com/gollvm/ project - it has all the above libraries among it's dependencies.
I am working on Fedora 33, under qemu.