riscv-software-src/riscv-tools #77: Error while building the tools

kllbrd, 2017-05-21:
[root@localhost riscv-tools]# ./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
libtoolize:   error: 'build-aux/config.guess' is newer: use '--force' to overwrite
libtoolize:   error: 'build-aux/config.sub' is newer: use '--force' to overwrite
Building project riscv-openocd
../../../src/target/riscv/riscv-013.c: In function ‘assert_reset’:
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1158:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1158:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1161:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 0);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1161:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 0);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c: In function ‘deassert_reset’:
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1186:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1186:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c: In function ‘riscv013_halt_current_hart’:
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1700:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 1);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1700:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 1);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1716:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 0);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1716:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 0);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c: In function ‘riscv013_reset_current_hart’:
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1856:13: note: in expansion of macro ‘set_field’
   control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
             ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1856:13: note: in expansion of macro ‘set_field’
   control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
             ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1864:13: note: in expansion of macro ‘set_field’
   control = set_field(control, DMI_DMCONTROL_HALTREQ, 0);
             ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1864:13: note: in expansion of macro ‘set_field’
   control = set_field(control, DMI_DMCONTROL_HALTREQ, 0);
             ^~~~~~~~~
../../../src/target/riscv/riscv-013.c: In function ‘riscv013_set_autoexec’:
../../../src/target/riscv/riscv-013.c:71:69: error: left shift of negative value [-Werror=shift-negative-value]
 #define get_field(reg, mask) (((reg) & (mask)) / ((mask) & ~((mask) << 1)))
                                                                     ^
../../../src/target/riscv/riscv-013.c:1982:21: note: in expansion of macro ‘get_field’
   uint32_t aa_aed = get_field(aa, DMI_ABSTRACTAUTO_AUTOEXECPROGBUF);
                     ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:72:85: error: left shift of negative value [-Werror=shift-negative-value]
 t_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                           ^
../../../src/target/riscv/riscv-013.c:1985:8: note: in expansion of macro ‘set_field’
   aa = set_field(aa, DMI_ABSTRACTAUTO_AUTOEXECPROGBUF, aa_aed);
        ^~~~~~~~~
cc1: all warnings being treated as errors
gmake[6]: *** [riscv/riscv-013.lo] Error 1
gmake[5]: *** [all-recursive] Error 1
gmake[4]: *** [all] Error 2
gmake[3]: *** [all-recursive] Error 1
gmake[2]: *** [all] Error 2
gmake[1]: *** [all-recursive] Error 1
gmake: *** [all] Error 2

any help would be appreciated, 
thanks

mwachs5, 2017-05-21:
What OS & version is this on?

kllbrd, 2017-05-22:
Thank you @mwachs5 ! 
The OS & Version : Fedora 25, GNOME Version 3.22.2.

jim-wilson, 2018-02-03:
This was apparently fixed by the change Palmer merged in.

closed 2018-02-03 by jim-wilson