riscv-software-src/riscv-tools #79: make -j16 ARCH=riscv errors

ninode, 2017-05-22:
Hello,
I'm having these issues building the Linux Kernel:

```
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/linux-3.14.33$ make ARCH=riscv defconfig
*** Default configuration is based on 'riscv64_spike'
#
# configuration written to .config
#

alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/linux-3.14.33$ make -j16 ARCH=riscv
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CC      scripts/mod/empty.o
gcc: error: unrecognized argument in option ‘-mabi=lp64’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized command line option ‘-mno-save-restore’
gcc: error: unrecognized command line option ‘-mstrict-align’
make[2]: *** [scripts/mod/empty.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC      scripts/mod/devicetable-offsets.s
gcc: error: unrecognized argument in option ‘-mabi=lp64’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized command line option ‘-mno-save-restore’
gcc: error: unrecognized command line option ‘-mstrict-align’
make[2]: *** [scripts/mod/devicetable-offsets.s] Error 1
make[1]: *** [scripts/mod] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [scripts] Error 2
make: *** Waiting for unfinished jobs....

```

If you could please tell me what I'm missing or how to fix it, I would very much appreciate it. 
Thank you !

kito-cheng, 2017-05-22:
Try add CROSS_COMPILE= riscv64-unknown-linux-gnu- to your command?

ninode, 2017-05-22:
kito-cheng, thank you for your reply. I tired what you've suggested, and I got the following error messages:

```
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/linux-3.14.33$ riscv64-unknown-linux-gnu-gcc --version
riscv64-unknown-linux-gnu-gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/linux-3.14.33$ make -j8 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CC      scripts/mod/empty.o
  CC      scripts/mod/devicetable-offsets.s
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-mstrict-align'; did you mean '-Wstrict-aliasing'?
make[2]: *** [scripts/mod/empty.o] Error 1
make[2]: *** Waiting for unfinished jobs....
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-mstrict-align'; did you mean '-Wstrict-aliasing'?
make[2]: *** [scripts/mod/devicetable-offsets.s] Error 1
make[1]: *** [scripts/mod] Error 2
make: *** [scripts] Error 2
make: *** Waiting for unfinished jobs....
```

palmer-dabbelt, 2017-05-22:
Your compiler is too old.  We recently added the -mstrict-align argument to GCC.

ninode, 2017-05-22:
Thank you Palmer. How can I upgrade the riscv64-unknown-linux-gnu-gcc ?

palmer-dabbelt, 2017-05-22:
The latest master from http://github.com/riscv/riscv-gnu-toolchain should have it.  Here's the actual patch

https://github.com/riscv/riscv-gcc/commit/2fe94d5a5a860a0226373397e05c20a873b77b4d

ninode, 2017-05-23:
I built the new repo with the newest version:
```
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip$ riscv64-unknown-elf-gcc  --version
riscv64-unknown-elf-gcc (GCC) 7.1.1 20170509
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip$ riscv64-unknown-linux-gnu-gcc --version
riscv64-unknown-linux-gnu-gcc (GCC) 7.1.1 20170509
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Then cross-compiled:  make -j8 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-
Issue solved. Thank you very much.

closed 2017-05-23 by ninode