riscv-software-src/riscv-tools #87: Linux sees only single core although Spike simulates two

MaMazav, 2017-06-04:
I'm trying to simulate multicore RISC-V processor on Spike using a "-p2" flag.
While debugging I do see that two processor_t objects are created.

However according to the following indications it seems that only single core is actually used:

1. I'm running simple vmlinux with Busybox according to the instructions in the README, and having the following cpuinfo content:

        / # cat /proc/cpuinfo 
        hart	: 0
        isa	: RV64G

2. When trying to execute the program below, I see by debugging Spike that all the program's instructions are executed on same processor.

Maybe Busybox doesn't utilize by default the two processor? Something else?
As first stage, also executing multithreading on the embedded simulator (without linux) is also applicable. I just don't know how to do that - pthread API is not available on the non-linux compiler and I wonder which threading API does.

The program mentioned above:

        #define _GNU_SOURCE

        #include<stdio.h>
        #include<string.h>
        #include<pthread.h>
        #include<stdlib.h>
        #include<unistd.h>

        int threads_count;
        cpu_set_t *cpuset;
        pthread_t *tid;

        int counter;
        pthread_mutex_t lock;

        void* doSomeThing(void *arg)
        {
            unsigned long i = 0;
            long index = (long)arg;

            int err = pthread_setaffinity_np(tid[index], 1, cpuset + index);
            //int err = pthread_setaffinity_np(pthread_self(), 1, cpuset + index);
            if (err != 0)
              printf("\ncan't INTERNALLY set affinity of thread %d: [%s]", index, strerror(err));

            printf("\n Job %d started\n", counter);

            for(i=0; i<10;i++)
            {
                pthread_mutex_lock(&lock);
        	counter++;
        	pthread_mutex_unlock(&lock);
            }

            printf("\n Job %d finished\n", counter);

            return NULL;
        }

        int main(char **argv, int argc)
        {
            int i = 0;
            int err;
            counter = 0;
            
            threads_count = 6;
            cpuset = (cpu_set_t *)malloc(sizeof(cpu_set_t) * threads_count);
            
            if (pthread_mutex_init(&lock, NULL) != 0)
            {
                printf("\n mutex init failed\n");
                return 1;
            }
    

            for (i = 0; i < threads_count; ++i)
            {
                CPU_ZERO(cpuset + i);
                CPU_SET(i, cpuset + i);
            }

            tid = (pthread_t *)malloc(sizeof(pthread_t) * threads_count);
    
            for (i = 0; i < threads_count; ++i)
            {
                long l = i;
                err = pthread_create(tid + i, NULL, &doSomeThing, (void *)l);
                if (err != 0)
                    printf("\ncan't create thread :[%s]", strerror(err));
	
        	err = pthread_setaffinity_np(tid[i], 1, cpuset + i);
	        if (err != 0)
        	    printf("\ncan't set affinity of thread %d: [%s]", i, strerror(err));
            }

            pthread_join(tid[0], NULL);
            pthread_join(tid[1], NULL);
            pthread_mutex_destroy(&lock);
    
            free(tid);
            free(cpuset);

            return 0;
        }

ben-k, 2017-06-05:
You may need to enable SMP support in Linux - I don't believe it's turned on by default.  Using the usual step from `riscv-linux`:

```
make ARCH=riscv menuconfig
```

And then under "Platform type" turn on "Symmetric Multi-Processing".

closed 2017-06-06 by MaMazav

beatsnbytes, 2018-11-27:
A year after....did this experiment worked @MaMazav  ?