riscv-software-src/riscv-tools #98: Missing tools

noamc, 2017-06-22:
During my riscv evaluation I found that I miss some well known projects I am sure will be needed by others too.
1) Buildroot
2) uClibc
3) u-boot

Is there a plan to add them to https://github.com/riscv/ ?
And later to maintain and upstream the riscv port?

Thanks
Noam

palmer-dabbelt, 2017-06-22:
In general we've been offering a repository on github.com/riscv to anyone who's in the process of producing a port and wants a place to synchronize their efforts to get the port upstream.  We would eventually like to support all of software on RISC-V, but there's a lot of software so it'll take a while.

* I know there are handful of buildroot ports floating around, but I don't think anything official or fully featured will show up until we're upstream in all the core system projects (Linux and glibc remain).
* I haven't head of a uclibc port.  We've been using newlib instead, which will be submitted upstream (though not by me).
* There was a u-boot port posted to the mailing list a while ago, but I know almost nothing about it, sorry.

The sw-dev mailing list might be a better place to ask about these ports, as this is more of a discussion than a bug in riscv-tools.

jim-wilson, 2018-02-03:
Answered by Palmer.

closed 2018-02-03 by jim-wilson