riscv-software-src/riscv-tools #99: Build Script Fails

Jmmanin, 2017-06-22:
Hello, I've been trying to install riscv-tools for a day or so without any luck. I'm working on Ubuntu 16.04 running on a Macbook using Parallels. Here is the output from the script:

`Starting RISC-V Toolchain build process

libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:48: installing 'build-aux/ar-lib'
configure.ac:43: installing 'build-aux/compile'
configure.ac:34: installing 'build-aux/config.guess'
configure.ac:34: installing 'build-aux/config.sub'
configure.ac:36: installing 'build-aux/install-sh'
configure.ac:36: installing 'build-aux/missing'
libjaylink/Makefile.am: installing 'build-aux/depcomp'
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
libtoolize: 'AC_PROG_RANLIB' is rendered obsolete by 'LT_INIT'
configure.ac:26: installing './compile'
configure.ac:37: installing './config.guess'
configure.ac:37: installing './config.sub'
configure.ac:16: installing './install-sh'
configure.ac:16: installing './missing'
Makefile.am:23: warning: wildcard $(srcdir: non-POSIX variable name
Makefile.am:23: (probably a GNU make extension)
Makefile.am: installing './INSTALL'
doc/Makefile.am:1: installing 'doc/mdate-sh'
doc/Makefile.am:1: installing 'doc/texinfo.tex'
src/Makefile.am: installing './depcomp'
Configuring project riscv-openocd
libtoolize:   error: 'build-aux/install-sh' is newer: use '--force' to overwrite
Building project riscv-openocd
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
fatal: No tags can describe '459b39ec67bbbe2ea0a192344c5236518da79c65'.
Try --always, or create some tags.
fatal: No tags can describe '459b39ec67bbbe2ea0a192344c5236518da79c65'.
Try --always, or create some tags.
ar: `u' modifier ignored since `D' is the default (see `U')
Installing project riscv-openocd
/bin/mkdir: cannot create directory ‘/riscv’: Permission denied
make[7]: *** [install-dist_ulinkDATA] Error 1
make[6]: *** [install-am] Error 2
make[5]: *** [install-recursive] Error 1
make[4]: *** [install-recursive] Error 1
make[3]: *** [install] Error 2
make[2]: *** [install-recursive] Error 1
make[1]: *** [install] Error 2
make: *** [install-recursive] Error 1`

I don't know if maybe I skipped a step or I'm missing something the installation instructions didn't mention (this is a pretty fresh install of Ubuntu) but I'm not really sure what's going on here.

closed 2017-06-23 by Jmmanin

agent001, 2017-06-27:
Similar issue
/rocket-chip/riscv-tools# date -u
Tue Jun 27 03:27:36 UTC 2017
/rocket-chip/riscv-tools# git submodule update --init --recursive
/rocket-chip/riscv-tools# . build.common
/rocket-chip/riscv-tools# build_project riscv-openocd --prefix=$RISCV --enable-remote-bitbang --enable-jtag_vpi --disable-werror

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
libtoolize:   error: 'build-aux/install-sh' is newer: use '--force' to overwrite
Building project riscv-openocd
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ln: failed to create hard link '.libs/libjtag.lax/lt1-core.o' => '.libs/libjtag.lax/libocdjtagdrivers.a/core.o': Operation not permitted
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ln: failed to create hard link '.libs/libflash.lax/lt1-core.o' => '.libs/libflash.lax/libocdflashnand.a/core.o': Operation not permitted
ln: failed to create hard link '.libs/libflash.lax/lt2-tcl.o' => '.libs/libflash.lax/libocdflashnand.a/tcl.o': Operation not permitted
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
fatal: No tags can describe '459b39ec67bbbe2ea0a192344c5236518da79c65'.
Try --always, or create some tags.
fatal: No tags can describe '459b39ec67bbbe2ea0a192344c5236518da79c65'.
Try --always, or create some tags.
ln: failed to create hard link '.libs/libopenocd.lax/lt1-transport.o' => '.libs/libopenocd.lax/libtransport.a/transport.o': Operation not permitted
ln: failed to create hard link '.libs/libopenocd.lax/lt2-core.o' => '.libs/libopenocd.lax/libflash.a/core.o': Operation not permitted
ln: failed to create hard link '.libs/libopenocd.lax/lt3-fileio.o' => '.libs/libopenocd.lax/libflash.a/fileio.o': Operation not permitted
ln: failed to create hard link '.libs/libopenocd.lax/lt4-lt1-core.o' => '.libs/libopenocd.lax/libflash.a/lt1-core.o': Operation not permitted
ln: failed to create hard link '.libs/libopenocd.lax/lt5-tcl.o' => '.libs/libopenocd.lax/libflash.a/tcl.o': Operation not permitted
ln: failed to create hard link '.libs/libopenocd.lax/lt6-target.o' => '.libs/libopenocd.lax/libtarget.a/target.o': Operation not permitted
ar: `u' modifier ignored since `D' is the default (see `U')
Installing project riscv-openocd
/rocket-chip/riscv-tools#

yuz835, 2017-07-10:
why u closed the issue. did u fixed it?

DarkLinkXXXX, 2018-02-18:
[image]

Jmmanin, 2018-02-18:
OP here, sheepishly coming out from the shadows. If I remember correctly, the error I got was becuase I had forgotten to sudo the build command. It was a stupid error so I kind of just swept it under the rug. I also remember having an issue with not having the libusb dependency installed but that’s since been added the to the apt-get statement in the readme. Sorry for up and disappearing.

sorear, 2018-02-19:
This repo is kinda unloved, most people use riscv-gnu-toolchain or freedom-u-sdk.

ksarkar82, 2019-03-31:
Hi All,
I've been trying to install riscv-tools for a day or so without any luck. I'm working on Ubuntu 16.04 running in my laptop ubuntu 18.04. Here is the output from the script:

koushik@koushik-Presario-CQ43-Notebook-PC:~/rocket/rocket-chip/riscv-tools$ ./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
ar: `u' modifier ignored since `D' is the default (see `U')
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
ar: `u' modifier ignored since `D' is the default (see `U')
Installing project riscv-openocd
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
/bin/mkdir: /bin/mkdir: cannot create directory ‘/rocket’: Permission denied
make[3]: *** [install-dist_ulinkDATA] Error 1
make[3]: *** Waiting for unfinished jobs....
cannot create directory ‘/rocket’: Permission denied
make[3]: *** [install-info-am] Error 1
make[2]: *** [install-am] Error 2
make[1]: *** [install-recursive] Error 1
make: *** [install] Error 2

 I'm not really sure what's going on here ,how to solve this !!! Kindly help ASAP, I am kinda stuck

with regards,
Koushik

jim-wilson, 2019-03-31:
You set the environment variable RISCV to a bad value.  This must be a directory that you have write access to, and it must be a value that works in both the shell and in make.  Absolute pathnames work best and should always work.  Something like RISCV=~/rocket may or may not work depending on your system and could result in an error like this.  If you are trying to do something like this, then use an absolute pathname instead.

jim-wilson, 2019-03-31:
You can get more info about what is going wrong if you look at riscv-openocd/build/build.log.

ksarkar82, 2019-04-01:
Thanks Jim,looks like it's working ,thank you so much

ksarkar82, 2019-04-01:
Hi Jim.
Now I am facing this --
In file included from /home/koushik/rocket/rocket-chip/riscv-tools/riscv-gnu-toolchain/riscv-gcc/libstdc++-v3/include/precompiled/extc++.h:49:0:
/home/koushik/rocket/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/build-gcc-newlib-stage2/riscv64-unknown-elf/libstdc++-v3/include/ext/extptr_allocator.h:41:10: fatal error: ext/pointer.h: No such file or directory
 #include <ext/pointer.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [riscv64-unknown-elf/bits/extc++.h.gch/O2g.gch] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all-target-libstdc++-v3] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-gcc-newlib-stage2] Error 2

-------
With farther check I observed pointer.h file "Link (broken) (inode/symlink)"...I do not know how many such files will be like that ,how to fix this ?
As you said I kept as absolute path "koushik@koushik-Presario-CQ43-Notebook-PC:~/rocket/rocket-chip/riscv-tools$ export RISCV=/home/koushik/riscvtools

with regards,
Koushik

jim-wilson, 2019-04-01:
Looks like the "Link (broken) (Inode/symlink)" messge is what you get from the GNOME Files application as the type.  The interesting part is actually the link target.  What is the target, and why doesn't it exist?  You probably made a mistake somewhere along the way.  If you did something like change the source/build directory name, then you might need to restart from the beginning to get a consistent build.

ksarkar82, 2019-04-01:
hi jim,
able to build now finally riscv tools, & emulator too, but now getting below error, dont know where need change & what to change to make the run-asm-tests.

koushik@koushik-Presario-CQ43-Notebook-PC:~/rocket/rocket-chip/emulator$ make -j2 run-asm-tests
make: *** No rule to make target '/home/koushik/riscvtools/riscv64-unknown-elf/share/riscv-tests/isa/rv64um-v-mul', needed by 'output/rv64um-v-mul'.  Stop.

jim-wilson, 2019-04-01:
I don't know anything about rocket-chip.  run-asm-tests looks like a rocket-chip question.

The target file /home/koushik/riscvtools/riscv64-unknown-elf/share/riscv-tests/isa/rv64um-v-mul should have been created by the riscv-tools build.  Try looking for the file, and looking at build logs, to see where it is, or why it might not be there.  If you have a confused build, you might need to restart from the beginning to get a clean build and install.

ksarkar82, 2019-04-19:
hi jim,
thank you for your support and guidance ...now i am facing issue like below::

koushik@koushik-Presario-CQ43-Notebook-PC:~/riscv_ks/rocket-chip/emulator$ makerm -rf verilator/src/verilator-4.008/
mkdir -p verilator/src/verilator-4.008/
cat verilator/verilator-4.008.tar.gz | tar -xz --strip-components=1 -C verilator/src/verilator-4.008/
### 
**gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now**
Makefrag-verilator:42: recipe for target 'verilator/src/verilator-4.008/configure' failed
make: *** [verilator/src/verilator-4.008/configure] Error 2

jim-wilson, 2019-04-19:
This isn't a riscv-tools problem, this is a rocket-chip problem, and I'm not a rocket-chip expert.  It looks like your .tar.gz file may be corrupted.  You could try checking it with "tar ztf ..." and if you get an error it is corrupted.