riscv-software-src/riscv-tools #101: Error building linux-4.6.2 with riscv-tools

Disasm, 2017-07-02:
I'm trying to build toolchain and Linux kernel for priv-1.9.1
Software versions:
- Host OS: Ubuntu 16.04 x86_64 inside Docker
- riscv-tools commit ad9ebb8557e32241bfca047f2bc628a2bc1c18cb (just as [riscv-qemu](https://github.com/riscv/riscv-qemu/) recommends)
- linux-4.6.2 with [riscv-linux branch priv-1.9](https://github.com/riscv/riscv-linux/tree/priv-1.9)

I build riscv64-unknown-linux-gnu version of the toolchain:
```
# mkdir riscv-tools/riscv-gnu-toolchain/build-linux
# cd riscv-tools/riscv-gnu-toolchain/build-linux
# ../configure --prefix=$RISCV --enable-linux
# make -j16
```
(kernel headers are already installed)

Everything builds as expected, but when I trying to build linux kernel I get the following errors:
```
riscv64-unknown-linux-gnu-ld: arch/riscv/kernel/entry.o: can't link hard-float modules with soft-float modules
riscv64-unknown-linux-gnu-ld: failed to merge target specific data of file arch/riscv/kernel/entry.o
riscv64-unknown-linux-gnu-ld: arch/riscv/kernel/sbi.o: can't link hard-float modules with soft-float modules
riscv64-unknown-linux-gnu-ld: failed to merge target specific data of file arch/riscv/kernel/sbi.o
riscv64-unknown-linux-gnu-ld: arch/riscv/kernel/vdso/built-in.o: can't link hard-float modules with soft-float modules
riscv64-unknown-linux-gnu-ld: failed to merge target specific data of file arch/riscv/kernel/vdso/built-in.o
```
Looks like this linker can't link "C" object files with "assembler" object files for some reason.

kllbrd, 2017-08-02:
Try `make -j16 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-`
@Disasm

jim-wilson, 2018-02-03:
The riscv-linux tree you used is unfortunately no longer available.

I checked out and built the riscv-gnu-toolchain tree and it appears to be working correctly for me.  The compiler normally passes -march and -mabi options to the assembler, but the defaults it is configured for should be correct, and it should still work correctly even without these options.

The hard-float abi, e.g. lp64d, is not link compatible with the soft-float abi, e.g. lp64.  You need to figure out which file was compiled/assembled wrong, and then figure out why.  You can see what an object file was compiled for by looking at the ELF header flags.  E.g.
gamma02:2117$ riscv64-unknown-linux-gnu-readelf -h cpu.o | grep Flags
  Flags:                             0x1, RVC, soft-float ABI

It looks like kernel files are all supposed to be compiled soft-float.  In current linux sources, I see that linux/arch/riscv/Makefile sets KBUILD_CFLAGS and KBUILD_AFLAGS to have both -march and -mabi options, which should and does give the correct results.  It is possible that there was a problem with older kernel sources.

closed 2018-02-03 by jim-wilson