riscv-software-src/riscv-tools #104: Error while building -j4 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- vmlinux

kllbrd, 2017-07-13:
Hello, 
I am getting this:

[root@localhost linux-3.14.33]# make -j4 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- vmlinux
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  GEN     usr/initramfs_data.cpio.gz
  CHK     include/generated/compile.h
  CC      kernel/sysctl.o
  CC      lib/lz4/lz4_decompress.o
  CC      drivers/of/fdt.o
  LD      kernel/built-in.o
  LD      lib/lz4/built-in.o
  CC      lib/lzo/lzo1x_decompress_safe.o
  CC      lib/xz/xz_dec_syms.o
  CC      lib/xz/xz_dec_stream.o
  LD      lib/lzo/lzo_decompress.o
  LD      lib/lzo/built-in.o
  CC      lib/zlib_inflate/inffast.o
  CC      lib/zlib_inflate/inflate.o
  LD      drivers/of/built-in.o
  CC      lib/xz/xz_dec_lzma2.o
  CC      lib/zlib_inflate/infutil.o
  LD      drivers/built-in.o
  CC      lib/zlib_inflate/inftrees.o
  CC      lib/xz/xz_dec_bcj.o
  CC      lib/zlib_inflate/inflate_syms.o
  LD      lib/zlib_inflate/zlib_inflate.o
  LD      lib/zlib_inflate/built-in.o
  CC      lib/decompress.o
  LD      lib/xz/xz_dec.o
  LD      lib/xz/built-in.o
  CC      lib/decompress_bunzip2.o
  CC      lib/decompress_inflate.o
  CC      lib/decompress_unlz4.o
  CC      lib/decompress_unlzma.o
  CC      lib/decompress_unlzo.o
  CC      lib/decompress_unxz.o
  AR      lib/lib.a
  EXPORTS lib/lib-ksyms.o
  LD      lib/built-in.o
  AS      usr/initramfs_data.o
  LD      usr/built-in.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  LD      init/built-in.o
  LD      vmlinux.o
  MODPOST vmlinux.o
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: warning: dot moved backwards before `.exit.text'
riscv64-unknown-linux-gnu-ld: section .exit.text LMA [000000005012cd58,000000005012cfd7] overlaps section .init.data LMA [0000000000014000,00000000d012cd57]
arch/riscv/kernel/head.o: In function `.L0 ':
(.init.text+0xc): relocation truncated to fit: R_RISCV_PCREL_HI20 against symbol `hart_lottery' defined in .sbss section in arch/riscv/kernel/built-in.o
riscv64-unknown-linux-gnu-ld: final link failed: Invalid operation
Makefile:1005: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1

Any idea why is this happening?
Thank you very much!

palmer-dabbelt, 2017-07-13:
* What Linux commit/branch and config are you using?
* What toolchain version are you using?

kllbrd, 2017-07-18:
Hello @palmer-dabbelt and sorry for the delay,

My linux commit/branches are all masters. I am not sure what do you mean by config??
My toolchain's version is: f5fae1c

Thank you!

kllbrd, 2017-07-18:
The rootfs.cpio had errors. Problem solved, thank you again.