riscv-software-src/riscv-tools #107: When install riscv-linux environment, tar exit with failure and make ARCH=riscv error

BenjaminLpn, 2017-07-23:
During these days, I have tried to install riscv environment in Ubuntu System. According to the page, https://github.com/riscv/riscv-tools, I installed the toolchain successfully.

Then I tried to build the Linux Kernel, using this way:
`$ cd $TOP
$ git clone https://github.com/riscv/riscv-linux.git linux-3.14.33
$ curl -L https://www.kernel.org/pub/linux/kernel/v3.x/linux-3.14.33.tar.xz | tar -xJkf -`
When I execute this line, tar exists with failure, like this:
`tar: linux-3.14.33/virt/kvm/arm/vgic.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/assigned-dev.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/async_pf.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/async_pf.h: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/coalesced_mmio.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/coalesced_mmio.h: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/eventfd.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/ioapic.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/ioapic.h: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/iodev.h: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/iommu.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/irq_comm.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/irqchip.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/kvm_main.c: Cannot open: File exists
tar: linux-3.14.33/virt/kvm/vfio.c: Cannot open: File exists
tar: Exiting with failure status due to previous errors`

I ignored this failure, and executed this:
`$ cd linux-3.14.33
$ make ARCH=riscv defconfig
$ make ARCH=riscv menuconfig
$ make -j8 ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-`
Then, some errors appeared:
`crypto/kpp.c: In function 'crypto_kpp_report':
crypto/kpp.c:31:27: error: storage size of 'rkpp' isn't known
  struct crypto_report_kpp rkpp;
                           ^~~~
crypto/kpp.c:35:19: error: 'CRYPTOCFGA_REPORT_KPP' undeclared (first use in this function); did you mean 'CRYPTOCFGA_REPORT_RNG'?
  if (nla_put(skb, CRYPTOCFGA_REPORT_KPP,
                   ^~~~~~~~~~~~~~~~~~~~~
                   CRYPTOCFGA_REPORT_RNG
crypto/kpp.c:35:19: note: each undeclared identifier is reported only once for each function it appears in
crypto/kpp.c:36:14: error: invalid application of 'sizeof' to incomplete type 'struct crypto_report_kpp'
       sizeof(struct crypto_report_kpp), &rkpp))
              ^~~~~~
crypto/kpp.c:31:27: warning: unused variable 'rkpp' [-Wunused-variable]
  struct crypto_report_kpp rkpp;
                           ^~~~
crypto/akcipher.c: In function 'crypto_akcipher_report':
crypto/akcipher.c:31:32: error: storage size of 'rakcipher' isn't known
  struct crypto_report_akcipher rakcipher;
                                ^~~~~~~~~
crypto/akcipher.c:35:19: error: 'CRYPTOCFGA_REPORT_AKCIPHER' undeclared (first use in this function); did you mean 'CRYPTOCFGA_REPORT_CIPHER'?
  if (nla_put(skb, CRYPTOCFGA_REPORT_AKCIPHER,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
                   CRYPTOCFGA_REPORT_CIPHER
crypto/akcipher.c:35:19: note: each undeclared identifier is reported only once for each function it appears in
crypto/akcipher.c:36:14: error: invalid application of 'sizeof' to incomplete type 'struct crypto_report_akcipher'
       sizeof(struct crypto_report_akcipher), &rakcipher))
              ^~~~~~
crypto/akcipher.c:31:32: warning: unused variable 'rakcipher' [-Wunused-variable]
  struct crypto_report_akcipher rakcipher;
                                ^~~~~~~~~
scripts/Makefile.build:302: recipe for target 'crypto/kpp.o' failed
make[1]: *** [crypto/kpp.o] Error 1
make[1]: *** Waiting for unfinished jobs....
scripts/Makefile.build:302: recipe for target 'crypto/akcipher.o' failed
make[1]: *** [crypto/akcipher.o] Error 1
crypto/acompress.c: In function 'crypto_acomp_report':
crypto/acompress.c:34:29: error: storage size of 'racomp' isn't known
  struct crypto_report_acomp racomp;
                             ^~~~~~
crypto/acompress.c:38:19: error: 'CRYPTOCFGA_REPORT_ACOMP' undeclared (first use in this function); did you mean 'CRYPTOCFGA_REPORT_AEAD'?
  if (nla_put(skb, CRYPTOCFGA_REPORT_ACOMP,
                   ^~~~~~~~~~~~~~~~~~~~~~~
                   CRYPTOCFGA_REPORT_AEAD
crypto/acompress.c:38:19: note: each undeclared identifier is reported only once for each function it appears in
crypto/acompress.c:39:14: error: invalid application of 'sizeof' to incomplete type 'struct crypto_report_acomp'
       sizeof(struct crypto_report_acomp), &racomp))
              ^~~~~~
crypto/acompress.c:34:29: warning: unused variable 'racomp' [-Wunused-variable]
  struct crypto_report_acomp racomp;
                             ^~~~~~
scripts/Makefile.build:302: recipe for target 'crypto/acompress.o' failed`

Thanks for your attention, and looking forward to your reply.

palmer-dabbelt, 2017-07-24:
Sorry, I'm not sure what's going on here.  You could try the "riscv-for-submission-v6" branch, which contains the full Linux source tree and therefor doesn't require you to run "tar":

````
$ git clone git://github.com/riscv/riscv-linux.git -b riscv-for-submission-v6
$ cd riscv-linux
$ make ARCH=riscv defconfig
$ make ARCH=riscv
````

BenjaminLpn, 2017-08-03:
@palmer-dabbelt 
Sorry for late answer, and sorry for new error:

  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  AR      init/built-in.o
  AR      built-in.o
  LD      vmlinux.o
  MODPOST vmlinux.o
drivers/pci/probe.o: In function `.L603':
probe.c:(.text+0x2d3c): undefined reference to `pcibios_fixup_bus'
drivers/pci/setup-res.o: In function `_pci_assign_resource':
setup-res.c:(.text+0x244): undefined reference to `pcibios_align_resource'
setup-res.c:(.text+0x250): undefined reference to `pcibios_align_resource'
setup-res.c:(.text+0x2a8): undefined reference to `pcibios_align_resource'
setup-res.c:(.text+0x2e0): undefined reference to `pcibios_align_resource'
Makefile:1001: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1

Thanks for your advice!

BenjaminLpn, 2017-08-08:
I clone the latest version. Everything is OK!

closed 2017-08-08 by BenjaminLpn