riscv-software-src/riscv-tools #109: -/bin/ash: not found error

Seonghun94, 2017-08-02:
I executed the instruction : `spike bbl vmlinux`

> [    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.12.0-rc2-677647-g1cd2e07 (seonghun@ubuntu) (gcc version 7.1.1 20170509 (GCC) ) #5 Tue Aug 1 21:41:51 PDT 2017
[    0.000000] Initial ramdisk at: 0xffffffff80018a48 (383572 bytes)
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2059588K/2095104K available (2360K kernel code, 141K rwdata, 458K rodata, 476K init, 246K bss, 35516K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=40000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] devtmpfs: initialized
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.000000] futex hash table entries: 256 (order: 0, 6144 bytes)
[    0.000000] NET: Registered protocol family 16
[    0.000000] vgaarb: loaded
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.000000] NET: Registered protocol family 2
[    0.000000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] TCP bind hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] TCP: Hash tables configured (established 16384 bind 16384)
[    0.000000] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.000000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.000000] NET: Registered protocol family 1
[    0.012000] Unpacking initramfs...
[    0.048000] console [sbi_console0] enabled
[    0.048000] workingset: timestamp_bits=62 max_order=19 bucket_order=0
[    0.064000] jitterentropy: Initialization failed with host not compliant with requirements: 2
[    0.064000] io scheduler noop registered
[    0.064000] io scheduler cfq registered (default)
[    0.064000] io scheduler mq-deadline registered
[    0.064000] io scheduler kyber registered
[    0.072000] Freeing unused kernel memory: 476K
[    0.072000] This architecture does not have kernel memory protection.
BusyBox v1.26.2 (2017-08-01 20:29:42 PDT) built-in shell (ash)

but i can't running instruction in busybox
`# ls`
> ls
   -/bin/ash:  ls: not found

`# echo`
> echo
    -/bin/ash:  echo: not found

kllbrd, 2017-08-02:
You have to install these binaries from busybox's menuconfig in order to use them.
I think by selecting applets.
@Seonghun94

Seonghun94, 2017-08-03:
Thanks @kllbrd
the busybox menuconfig setting list is...

> CONFIG_STATIC=y, listed as "Build BusyBox as a static binary (no shared libs)" in BusyBox Settings → Build Options
CONFIG_CROSS_COMPILER_PREFIX=riscv64-unknown-linux-gnu-, listed as "Cross Compiler prefix" in BusyBox Settings → Build Options
CONFIG_FEATURE_INSTALLER=y, listed as "Support --install [-s] to install applet links at runtime" in BusyBox Settings → General Configuration
CONFIG_INIT=y, listed as "init" in Init utilities
CONFIG_ASH=y, listed as "ash" in Shells
CONFIG_ASH_JOB_CONTROL=n, listed as "Ash → Job control" in Shells
CONFIG_MOUNT=y, listed as "mount" in Linux System Utilities
CONFIG_FEATURE_USE_INITTAB=y, listed as "Support reading an inittab file" in Init Utilities

is that right? then, I already done

kllbrd, 2017-08-03:
Yes, this is right. But in order to run binaries via busybox you have to open menuconfig, go to the applets section and select from there which binaries you want.
---Applets
    Coreutils  ---> ls (for example)
    ...etc.
@Seonghun94

Seonghun94, 2017-08-04:
@kllbrd Thank you very much!!!! It's running now. ^^

closed 2017-08-26 by palmer-dabbelt