riscv-software-src/riscv-tools #110: end Kernel panic - not syncing : no working init found

CheolWon, 2017-08-04:
I got error when i commanded 'spike bbl vmlinux'

[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.12.0-rc2-677647-g1cd2e07 (jcw@ubuntu) (gcc version 7.1.1 20170509 (GCC) ) #3 Tue Aug 1 19:41:16 PDT 2017
[    0.000000] Initial ramdisk at: 0xffffffff80019a18 (511755 bytes)
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 2059324K/2095104K available (2429K kernel code, 153K rwdata, 514K rodata, 604K init, 244K bss, 35780K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:0 nr_irqs:0 0
[    0.000000] riscv,cpu_intc,0: 64 local interrupts mapped
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 191126044627 ns
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 20.00 BogoMIPS (lpj=40000)
[    0.000000] pid_max: default: 32768 minimum: 301
[    0.000000] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] devtmpfs: initialized
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.000000] futex hash table entries: 256 (order: 0, 6144 bytes)
[    0.000000] NET: Registered protocol family 16
[    0.000000] vgaarb: loaded
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.000000] NET: Registered protocol family 2
[    0.000000] TCP established hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] TCP bind hash table entries: 16384 (order: 5, 131072 bytes)
[    0.000000] TCP: Hash tables configured (established 16384 bind 16384)
[    0.000000] UDP hash table entries: 1024 (order: 3, 32768 bytes)
[    0.000000] UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
[    0.000000] NET: Registered protocol family 1
[    0.020000] Unpacking initramfs...
[    0.064000] console [sbi_console0] enabled
[    0.064000] workingset: timestamp_bits=62 max_order=19 bucket_order=0
[    0.080000] jitterentropy: Initialization failed with host not compliant with requirements: 2
[    0.080000] io scheduler noop registered
[    0.080000] io scheduler cfq registered (default)
[    0.080000] io scheduler mq-deadline registered
[    0.080000] io scheduler kyber registered
[    0.088000] Freeing unused kernel memory: 604K
[    0.088000] This architecture does not have kernel memory protection.
[    0.092000] Failed to execute /init (error -8)
[    0.092000] Starting init: /sbin/init exists but couldn't execute it (error -8)
[    0.092000] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    0.092000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.12.0-rc2-677647-g1cd2e07 #3
[    0.092000] Call Trace:
[    0.092000] [<ffffffff80099414>] walk_stackframe+0x0/0xcc
[    0.092000] [<ffffffff802f2044>] rest_init+0x84/0x88
[    0.092000] [<ffffffff800e1d44>] panic+0xe0/0x200
[    0.092000] [<ffffffff802f2044>] rest_init+0x84/0x88
[    0.092000] [<ffffffff802f214c>] kernel_init+0x104/0x108
[    0.092000] [<ffffffff802f2044>] rest_init+0x84/0x88
[    0.092000] [<ffffffff80097da4>] ret_from_syscall+0xc/0x10
[    0.092000] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.

please help me, i want install riscv tools

palmer-dabbelt, 2017-08-04:
You probably have a dynamically linked /init but don't have /lib/libc.so.6 in your initramfs.

CheolWon, 2017-08-05:
@palmer-dabbelt 
Thank you for your answer :)
I checked and I have libc.so.6 $TOP/riscv/systroot/lib
Is there any other problems??..
I have not yet solved this problem

palmer-dabbelt, 2017-08-05:
You need to check if libc.so.6 is in your initramfs, not just your sysroot.  What is your initramfs config file?

CheolWon, 2017-08-05:
initramfs is in linux-3.14.33/include/config and there no libc.so.6

CheolWon, 2017-08-05:
@palmer-dabbelt 
Thank you for answering me!!
I solved my problem
I downloaded linux-3.14.33 and i changed linux-3.14.33 -> riscv-linux
then installed succesfully and it runs!!

jim-wilson, 2018-02-07:
Answered by Palmer.

closed 2018-02-07 by jim-wilson