riscv-software-src/riscv-tools #128: Error with emulator-rocketchip-BOOMConfig (cannot generate right *.out file)

BenjaminLpn, 2017-08-24:
Yesterday, I run Linux on FPGA with BOOM-core successfully. While I want to learn how the tool-chain works out, I surprisingly find that I cannot generate .out files.

Here is the output:
`
[lipn@lipn-P50 ~/Software/rocket-chip/emulator]make output/towers.riscv.out
mkdir -p ./output
ln -fs /home/lipn/Software/riscv/riscv-tools/riscv/riscv64-unknown-elf/share/riscv-tests/benchmarks/towers.riscv output/towers.riscv
./emulator-rocketchip-BOOMConfig +max-cycles=100000000 +verbose output/towers.riscv 3>&1 1>&2 2>&3 | /home/lipn/Software/riscv/riscv-tools/riscv/bin/spike-dasm  > output/towers.riscv.out && [ $PIPESTATUS -eq 0 ]
Makefile:40: recipe for target 'output/towers.riscv.out' failed
make: *** [output/towers.riscv.out] Error 1
[lipn@lipn-P50 ~/Software/rocket-chip/emulator]cat output/towers.riscv.out 
using random seed 1503591537
emulator-rocketchip-BOOMConfig: ../fesvr/dtm.cc:556: void dtm_t::producer_thread(): Assertion `get_field(hartinfo, DMI_HARTINFO_NSCRATCH) > 0' failed.
`

If there is any suggestion, please help me!
Thanks a lot!

BenjaminLpn, 2017-08-25:
I clone rocket-chip and riscv-tools again in another PC. And the same error appears.

ccelio, 2017-08-26:
It appears you have a mismatch between your version of BOOM and the version of riscv-tools that you built. 

Which riscv-tools did you build?

BenjaminLpn, 2017-08-26:
That could be the reason, I built the latest version of riscv-tools.
I am cloning the riscv-tools in the rocket-chip directory. 
If there is any new progress, I will inform you soon.
Thanks again.

BenjaminLpn, 2017-08-26:
It works out. Thanks!

closed 2017-08-26 by BenjaminLpn

manikantabhagavatula, 2017-12-08:
Hello @BenjaminLpn I am facing the same issue but I couldn't understand how to overcome it. Could you please help me with fixing it. Thank you.

BenjaminLpn, 2017-12-09:
@manikantabhagavatula There may be a mismatch between your version of BOOM and the version of riscv-tools. The convenient way is to  build the riscv-tools in your rocket-chip repo.  Good luck.

manikantabhagavatula, 2017-12-09:
@BenjaminLpn I have tried using the same version of BOOM and riscv-tools and it worked for me. Thanks alot. Also, is it possible to build riscv-tools necessary for both rocket (branch "master")and boom (branch "boom") on a single machine (within the same directory of /rocket-chip/riscv-tools)? To be more precise, Can I switch between rocket and boom using "git checkout master & git checkout boom" and compile programs simultaneously because I would like to compare the performance evaluation of both rocket and boom. Any input will be greatly appreciated.

BenjaminLpn, 2017-12-09:
@manikantabhagavatula Sure. As long as rocket and boom are switched in the same version, they can be compiled correctly with the same riscv-tools.