riscv-software-src/riscv-tools #132: R_RISCV_ALIGN

mamijaz, 2017-09-14:
When i compile a sample code bellow with fresh build riscv32-unknown-elf-gcc on Ubuntu 16.04

int main(){
	int a = 10;
	int b = 20;
	*(int*)0x07FC = a+b;

	while(1){}

	return 0;
}

i got the results as 

add.o:     file format elf32-littleriscv
Disassembly of section .text:
00000000 <main-0x3>:
   0:	00 01 00    	# R_RISCV_ALIGN
00000003 <main>:
   3:	fe010113          	addi	sp,sp,-32
   7:	00812e23          	sw	s0,28(sp)
   b:	02010413          	addi	s0,sp,32
   f:	00a00793          	li	a5,10
  13:	fef42623          	sw	a5,-20(s0)
  17:	01400793          	li	a5,20
  1b:	fef42423          	sw	a5,-24(s0)
  1f:	7fc00793          	li	a5,2044
  23:	fec42683          	lw	a3,-20(s0)
  27:	fe842703          	lw	a4,-24(s0)
  2b:	00e68733          	add	a4,a3,a4
  2f:	00e7a023          	sw	a4,0(a5)
00000033 <.L2>:
  33:	0000006f          	j	66 <.L2+0x33>

but why there is a "R_RISCV_ALIGN" , is this a new feature, if then how can i remove it, will it not generate a misalignment exception. few weeks ago i was getting something like 

add.o:     file format elf32-littleriscv
Disassembly of section .text:
00000000 <main>:
   0:	fe010113          	addi	sp,sp,-32
   4:	00812e23          	sw	s0,28(sp)
   8:	02010413          	addi	s0,sp,32
   c:	00a00793          	li	a5,10
  10:	fef42623          	sw	a5,-20(s0)
  14:	01400793          	li	a5,20
  18:	fef42423          	sw	a5,-24(s0)
  1c:	7fc00793          	li	a5,2044
  20:	fec42683          	lw	a3,-20(s0)
  24:	fe842703          	lw	a4,-24(s0)
  28:	00e68733          	add	a4,a3,a4
  2c:	00e7a023          	sw	a4,0(a5)
00000030 <.L2>:
  30:	0000006f          	j	30 <.L2>

closed 2017-09-14 by mamijaz

palmer-dabbelt, 2017-09-14:
FYI: This is eliminated by linker relaxation, there's a blog post about how it works.  https://www.sifive.com/blog/2017/08/28/all-aboard-part-3-linker-relaxation-in-riscv-toolchain/

kito-cheng, 2017-09-15:
Hi @palmer-dabbelt does it mean align will insert 3 byte by default now? but I remember the conclusion about this issue is it should work even linker not support relax with code shrink? https://github.com/riscv/riscv-binutils-gdb/pull/88 

cc: @pkmx

palmer-dabbelt, 2017-09-15:
> riscv/riscv-binutils-gdb#88

This is a regression, fixing it is on my TODO list.  Hopefully it'll be
done by the end of next week.

kito-cheng, 2017-09-15:
Oh, thanks, I just worry about lld can't support that in this moment :)

palmer-dabbelt, 2017-09-15:
Yep.  It's super awkward to break their stuff right after we told them it was fine and it would work...  sorry!

kito-cheng, 2017-09-15:
BTW,we've implemented similar method but always use nop16  to stuff that, it will work even linker does not shrinking the code.

palmer-dabbelt, 2017-09-15:
That's actually what I did upstream, this just leaked into a binary
release.  The problem is that approach still emits C instructions on non-C
targets.

Can I see your patch?

kito-cheng, 2017-09-15:
For non-C target, I thought it should not a problem since all instructions
is 32 bits? Hummm, but the problem that you want to resolve is when mixing
C and non-C code, I'll ask my colleague next week.