riscv-software-src/riscv-tools #134: build.sh is failing for riscv-openocd

akhileshian, 2017-09-25:
After cloning, I ran the ./build.sh, which is failing and showing the following error message, regarding missing **Makefile.in**:

$ ./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
configure: WARNING: libusb-1.x not found, trying legacy libusb-0.1 as a fallback; consider installing libusb-1.x instead
config.status: error: cannot find input file: `Makefile.in'

What could the error be?

Regards,
Akhilesh

brytonhall, 2017-10-03:
Is `libusb` installed on your system?

    ldconfig -p | grep libusb

masics, 2017-10-15:
I have the same problem and libusb is installed:

> $ /sbin/ldconfig -p | grep libusb
>         libusbredirparser.so.1 (libc6,x86-64) => /usr/lib64/libusbredirparser.so.1
>         libusbredirhost.so.1 (libc6,x86-64) => /usr/lib64/libusbredirhost.so.1
>         libusbpp-0.1.so.4 (libc6,x86-64) => /usr/lib64/libusbpp-0.1.so.4
>         libusbpp-0.1.so.4 (libc6) => /usr/lib/libusbpp-0.1.so.4
>         libusbmuxd.so.1 (libc6,x86-64) => /usr/lib64/libusbmuxd.so.1
>         libusb-1.0.so.0 (libc6,x86-64) => /usr/lib64/libusb-1.0.so.0
>         libusb-0.1.so.4 (libc6,x86-64) => /usr/lib64/libusb-0.1.so.4
>         libusb-0.1.so.4 (libc6) => /usr/lib/libusb-0.1.so.4
>

brytonhall, 2017-10-16:
Does `riscv-openocd/Makefile.in` exist? If not, remove `riscv-openocd/configure` and run `./build.sh` again (it will run `autoreconf` which will create the required `Makefile.in` for `configure` to use).

masics, 2017-10-17:
After removing configure file I have following errors:
configure.ac:37: the top level
src/Makefile.am:4: bad characters in variable name `%C%_openocd_SOURCES'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:7: bad characters in variable name `%C%_libopenocd_la_SOURCES'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:11: bad characters in variable name `%C%_openocd_LDADD'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:13: bad characters in variable name `%C%_openocd_LDADD'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:16: bad characters in variable name `%C%_openocd_LDADD'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:18: bad characters in variable name `%C%_openocd_LDADD'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:21: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:26: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:27: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:29: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:30: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:31: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:35: bad characters in variable name `%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:38: bad characters in variable name `%C%_libopenocd_la_LDFLAGS'
Makefile.am:143:   `src/Makefile.am' included from here
src/Makefile.am:46: bad characters in variable name `%C%_libopenocd_la_LIBADD'
Makefile.am:143:   `src/Makefile.am' included from here
doc/Makefile.am:2: bad characters in variable name `%C%_openocd_TEXINFOS'
Makefile.am:144:   `doc/Makefile.am' included from here
Makefile.am:46: wildcard $(srcdir: non-POSIX variable name
Makefile.am:46: (probably a GNU make extension)
automake: cannot open < ./%D%/openocd.texi: No such file or directory
autoreconf: automake failed with exit status: 1

brytonhall, 2017-10-17:
You should mark any output as code (that is, enclose it in three backticks: \`\`\`) to avoid having it parsed as Markdown.

The variable names in `riscv-openocd/src/Makefile.am` shouldn't end with an apostrophe; so, for example, check that line four reads:

    %C%_openocd_SOURCES = \

 and not

    %C%_openocd_SOURCES' = \

masics, 2017-10-17:
Sorry about that.
The lines 4-5 in `riscv-openocd/src/Makefile.am` have no apostrophe.
```
%C%_openocd_SOURCES = \
        %D%/main.c
```

brytonhall, 2017-10-17:
No problem; I think your version of `automake` is too old to know how to expand those variables. Which version are you using?

masics, 2017-10-17:
```
$ automake --version
automake (GNU automake) 1.11.1
```

brytonhall, 2017-10-17:
`%C%` is expanded to the absolute pathname of a fragment in an include directive ([docs](https://www.gnu.org/software/automake/manual/automake.html#Include))

But this feature seems to have been added in [1.14](https://lists.gnu.org/archive/html/info-gnu/2013-06/msg00009.html). Can you upgrade to at least that version?

masics, 2017-11-05:
I installed a new automake but this time I get:
`error: Libtool library used but 'LIBTOOL' is undefined`

brytonhall, 2017-11-10:
Is `libtool` installed on your system?

angrej2504, 2017-12-28:
Hi,

I am getting following errors 
[image]

could you please help me out.

timsifive, 2017-12-28:
Is pkg-config installed? That seems to be a common pitfall, though I don’t
recognize this failure mode.
​

frohic, 2017-12-31:
I'm on centos 7.4
Similar problems.  installed version of `automake `is 1.13.4
Anyone know how to update it to 1.14 or later?  `sudo yum update automake` indicates it's currently updated.

masics, 2017-12-31:
You have to compile it to a custom directory and point your PATH there.

Best regards,
Michael Strelnikov

frohic, 2018-01-01:
Thanks Michael, I tried that.  Following these instructions from [](https://techglimpse.com/install-update-autoconf-linux-tutorial/
```
$ wget http://ftp.gnu.org/gnu/automake/automake-1.14.tar.gz
$ tar xvzf automake-1.14.tar.gz
$ cd automake-1.14
$ ./configure
$ make
$ sudo make install

```
Unfortunately I now get the following error that I didn't used to get: 
```
> ./build.sh 
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
configure: WARNING: libusb-1.x not found, trying legacy libusb-0.1 as a fallback; consider installing libusb-1.x instead
config.status: error: cannot find input file: `Makefile.in'

```
Do you know how I can go back to the original version of automake?  I'm worried that I may have broken something.

A wider issue here is that hardware engineers working in industry use RedHat/CentOS and the riscv tools don't seem to be able to build on it, certainly in a straightforward way.

masics, 2018-01-02:
I told to compile into separate folder. Now you have to remove those files
and reinstall original RPM.

Best regards,
Michael Strelnikov

gajinderpanesar, 2018-02-06:
I had similar problem with building the tools.
So I downloaded the latest automake. In my case this is version 15.1. In order to make this work I had to do
./configure --prefix=/usr

otherwise I got lots of errors like:

a.	Configure.ac:204: error: possibly undefined macro: AC_DEFINE
b.	Configure.ac:204: error: possibly undefined macro: AC_MSG_WARN
c.	Configure.ac:204: error: possibly undefined macro: AC_MSG_NOTICE

I even tried:

ln –s of /usr/share/aclocal/*.m4 into /usr/local/share/aclocal/

and this did not work, well I got different but similar errors.
So in the end the only thing that worked was:

./configure --prefix=/usr

dogrush, 2018-05-31:
I also had the same problem, and the reason is the missed *.m4 files.

I installed the newest automake, autoconf, libtool, pkg-config.
And then linked the aclocal/*.m4 files from the install path of libtool and pkg-config into share/aclocal path of automake.
Also, in the automake installed path (for me, automake-1.15 is installed), there are two aclocal path in automake-1.15/share: automake-1.15/share/aclocal, and automake-1.15/share/aclocal-1.15. So, I have to link all *.m4 files from automake-1.15/share/aclocal-1.15 into automake-1.15/share/aclocal.

That is all what I have done to solve the involved problems.

willeggeredwin, 2018-08-23:
I was having the same problem (Centos 7) and was able to solve it with the following steps:

1.) **Change** into folder riscv-openocd of your riscv-tools and remove old **configure**
`rm configure`

2.) In my case some libraries were missing and I had to **install** the following ones: 
`sudo yum install openocd`
`sudo yum install libusb1`
`sudo yum install libusb-devel`
`sudo yum install hidap`

3.)  **Add** the path to autoreconf 
`autoreconf -i -I /usr/share/aclocal/`

4.) **Change** to folder of riscv-tools and **execute** build again:
`./build.sh`

After these steps I tested my environment and it works.

zhanggd, 2018-09-07:
How about this question? Could you guys successfully build the riscv tools under CentOS?

jim-wilson, 2018-09-07:
riscv-tools is poorly maintained and difficult to build.  I don't recommend using it.  We are planning to deprecate the package.  Use riscv-gnu-toolchain instead if you want a compiler.  Or sifive/freedom-u-sdk if you want to build  a linux+buildroot system and boot it on a simulator.  You only need openocd if you have embedded hardware like the hifive1 board and want to use gdb to talk to it.

openocd is very difficult to build.  I have built it under Centos, but had to provide my own autoconf/automake versions.  Note that openocd is not our package, so there isn't much we can do about build problems with the package.  If you really need openocd, it may be easier to switch to a different OS that openocd supports.  We use Ubuntu 16.04 inside SiFive currently, and it builds OK there.

willeggeredwin, 2018-09-07:
I was able to build it under CentOS 7.