riscv-software-src/riscv-tools #139: Compile/Assembler error regarding misuse of csrr and csrw instructions

sherrbc1, 2017-10-19:
The error I am observing is as follows:

```
../../Source/portable/GCC/RISCV/port.c: Assembler messages:
../../Source/portable/GCC/RISCV/port.c:135: Error: Instruction csrr requires absolute expression
../../Source/portable/GCC/RISCV/port.c:137: Error: Instruction csrw requires absolute expression
```

This same issue has been brought up a few times before:
https://github.com/riscv/riscv-tests/issues/9
https://github.com/riscv/riscv-tests/issues/13
https://stackoverflow.com/questions/29860388/riscv-gcc-4-9-2-cant-compile-linux-3-14-29

Where the problem is addressed as one in which the compiler and assembler are somewhat out of phase. In this case, it was mentioned that the assembler is not understanding the output of the compiler due to this mismatch. The solution was simply to ensure these two tools were in sync. 

I first encountered this problem when building the `master` branch of `riscv-tools`. Then, I tried building the `release` branch, but observed the same behavior when trying to compile this code:

```
__asm volatile("csrr t0,mtime");
__asm volatile("add t0,t0,%0"::"r"(configTICK_CLOCK_HZ / configTICK_RATE_HZ));
__asm volatile("csrw mtimecmp,t0"); 
```

mwachs5, 2017-10-19:
`mtime` and `mtimecmp` cannot be accessed with CSR instructions. They are memory-mapped locations. You will need to change your code to memory accesses not CSR accesses.

sherrbc1, 2017-10-19:
@mwachs5 I am quite new to RISC-V. Has this always been true? This particular code segment was lifted from a RISC-V port of FreeRTOS located [here](https://interactive.freertos.org/hc/en-us/community/posts/210030246-32-bit-and-64-bit-RISC-V-using-GCC). 

I was reading section 3.1.14 (and Table 2.5 on page 10 for a list of M-mode CSRs) of the [RISC-V Instruction Set Manual Vol II](https://people.eecs.berkeley.edu/~krste/papers/riscv-privileged-v1.9.pdf) and was under the impression that `mtime` was a Machine-mode CSR, while `mtimecmp` was memory mapped.

I am trying to get this FreeRTOS port running in the Spike emulator (for which is _claims_ to have been originally developed) but am struggling to find documentation regarding its memory map. If `mtime` and `mtimecmp` really are just memory-mapped, I assume Spike has them fixed at some location. Or does this in some way work differently on RISC-V than other architectures?

sherrbc1, 2017-10-19:
@mwachs5 I am quite new to RISC-V. Has this always been the case? This code was lifted from a community port to RISC-V of FreeRTOS located [here](https://interactive.freertos.org/hc/en-us/community/posts/210030246-32-bit-and-64-bit-RISC-V-using-GCC).

mwachs5, 2017-10-19:
That link is Priv Spec v1.9, which is out of date, I suggest you always use the one on riscv.org, currently at v1.10 (you are right, it was not "always this way", but if you are using the latest version of riscv-tools, it certainly is):

https://riscv.org/specifications/privileged-isa/

The FreeRTOS port you are using may have been developed targeting 1.9, which means it needs to be updated to 1.10 or you need to use 1.9 compatible versions of Spike and RISC-V Tools.

In the current Spike, mtime and mtimecmp are offsets within the CLINT peripheral:
https://github.com/riscv/riscv-isa-sim/blob/b3caeee9858b3417d4110b921305c441d3c87653/riscv/clint.cc

And the CLINT_BASE is defined in this file: https://github.com/riscv/riscv-isa-sim/blob/b3caeee9858b3417d4110b921305c441d3c87653/riscv/encoding.h

sherrbc1, 2017-11-02:
@mwachs5 This was indeed the problem. I was not aware that RISCV was moving so quickly.

closed 2017-11-02 by sherrbc1