riscv-software-src/riscv-tools #140: can't make vmlinux

baemyoungsub, 2017-10-20:
`
cd $TOP/riscv-tools
curl https://www.kernel.org/pub/linux/kernel/v3.x/linux-3.14.33.tar.xz \
  | tar -xJ
cd linux-3.14.33

git init
git remote add origin https://github.com/riscv/riscv-linux.git                          
git fetch                                               
git checkout -f 989153f

Note: checking out '989153f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 989153f... Enhance stack tracing infrastructure
bae@ubuntu:~/lowRISC/lowrisc-chip/riscv-tools/linux-3.14.13$ make ARCH=riscv defconfig 
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
*** Default configuration is based on 'riscv_defconfig'
#
# configuration written to .config
#
bae@ubuntu:~/lowRISC/lowrisc-chip/riscv-tools/linux-3.14.13$ sudo make ARCH=riscv –j vmlinux 
scripts/kconfig/conf --silentoldconfig Kconfig
make: riscv-linux-gcc: Command not found
make: *** No rule to make target `–j'.  Stop
`

after that, nothing happended...
I just want to see "vmlinux" file... but I tried this thing several times.
please. someone tell me what is wrong...

brytonhall, 2017-10-20:
The `-j` flag for `make` requires an argument; for example

    make ARCH=riscv -j4 vmlinux

(see `man make` for details) and you shouldn't be building software as the superuser.

seldridge, 2017-10-20:
You don't have `riscv-linux-gcc` on your path. You need to install the _linux_ version of the RISC-V toolchain first. In `riscv-tools` this is not handled automatically by the `build.sh` script. Refer to https://github.com/riscv/riscv-gnu-toolchain/blob/65cb174d37e2efe99e094eb317113442c79ed3ce/README.md#installation-linux for how to build this. If you've already done this, then you need to make sure that it's on your path, i.e., `which riscv-linux-gcc` should return a non-zero exit code.

Running `make -j` is fine---that causes `make` to run with as many parallel jobs as it can. The numeric argument is optional.

baemyoungsub, 2017-10-25:
Thanks! I had a lot of problems about the issue.  you gave me a key to solving it.