riscv-software-src/riscv-tools #148: Error building under the Ubunto subsystem of Windows 10

ComponentFactory, 2017-11-14:
After installing the Ubuntu subsystem on Windows 10, I would like to build the riscv-tools so I can compile simple C programs and see the generated assembly code. 

Following the instructions, the first few commands work fine:- 

```
    export TOP=$(pwd)
    sudo apt-get install git
    git clone https://github.com/riscv/riscv-tools.git
    cd $TOP/riscv-tools
    git submodule update --init --recursive
```

...but this one then fails...

```
    sudo apt-get install autoconf automake autotools-dev curl 
                 device-tree-compiler libmpc-dev libmpfr-dev libgmp-dev 
                 gawk build-essential bison flex texinfo gperf libtool 
                 patchutils bc
```

...with the following error...

```
    E: Package 'texinfo' has no installation candidate
    E: Unable to locate package gperf
```

..but it did suggest install texinfo-doc-nonfree, install-info and info instead so I did that and it installed them. So then I tried the remaining steps:-

```
    export RISCV=$TOP/riscv
    export PATH=$PATH:$RISCV/bin
    ./build.sh
```

...which failed with...

```
    Starting RISC-V Toolchain build process

    configure.ac:42: installing 'build-aux/ar-lib'
    configure.ac:37: installing 'build-aux/compile'
    configure.ac:28: installing 'build-aux/config.guess'
    configure.ac:28: installing 'build-aux/config.sub'
    configure.ac:30: installing 'build-aux/install-sh'
    configure.ac:30: installing 'build-aux/missing'
    Makefile.am: installing './INSTALL'
    libjaylink/Makefile.am:23: error: Libtool library used but 'LIBTOOL' is    undefined
    libjaylink/Makefile.am:23:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
    libjaylink/Makefile.am:23:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
    libjaylink/Makefile.am:23:   If 'LT_INIT' is in 'configure.ac', make sure
    libjaylink/Makefile.am:23:   its definition is in aclocal's search path.
    libjaylink/Makefile.am: installing 'build-aux/depcomp'
    autoreconf: automake failed with exit status: 1
```

..and now I am stuck. Anyone with an idea how to fix this?

evancox10, 2017-11-15:
What's your version of Windows and WSL? Are you on Ubuntu 16.04, or the older 14.04?

ComponentFactory, 2017-11-15:
Version: 16.04

0xSoEasY, 2020-11-15:
sudo apt-get install libtool