riscv-software-src/riscv-tools #152: couldn't run the hello world test

WaleedOsama, 2017-11-26:
i ran the build.sh script  , and i was ready to run the hello world test
so i ran this sequence as the README file 
'''
 cd $TOP
'''
'''
echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c
'''
'''
riscv64-unknown-elf-gcc -o hello hello.c
'''

'''
spike pk hello
'''

and this error appeared
'''
terminate called after throwing an instance of 'std::runtime_error'
  what():  couldn't allocate 2147483648 bytes of target memory
Aborted (core dumped)
'''

any help would be appreciated

WaleedOsama, 2017-11-27:
the problem was on my side , some unexpected things happened while i was installing the toolchain , however i tried it again with new virtual machine with bigger disk size and everything worked out. thank you

closed 2017-11-27 by WaleedOsama

rossberger, 2017-11-30:
I had this same issue when trying to run `spike pk hello` in vmware/Ubuntu 16.04.  Resolved by shutting down the vm, editing vm settings to increase RAM to >2GB.  After a restart spike ran successfully.