riscv-software-src/riscv-tools #163: Problems running make ARCH=riscv

pierceja, 2017-12-22:
Hello,
I am running into problems running make ARCH=riscv. When I run this, I get the console output below:

gcc: error: unrecognized argument in option ‘-mabi=lp64’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized command line option ‘-mno-save-restore’
Kbuild:21: recipe for target 'kernel/bounds.s' failed
make[1]: *** [kernel/bounds.s] Error 1
Makefile:1057: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2

I tried to configure linux (make ARCH=riscv menuconfig) to set the cross compiler prefix to riscv64-unknown-linux-gnu-gcc-. This gave me the console output below which made me think that was the wrong direction to go:

In file included from ./include/linux/syscalls.h:74:0,
                 from init/main.c:20:
./include/linux/signal.h:14:20: error: redefinition of 'copy_siginfo'
 static inline void copy_siginfo(struct siginfo *to, struct siginfo *from)
                    ^~~~~~~~~~~~
In file included from ./arch/riscv/include/uapi/asm/siginfo.h:22:0,
                 from ./include/uapi/linux/signal.h:6,
                 from ./include/linux/signal_types.h:10,
                 from ./include/linux/sched.h:27,
                 from ./include/linux/binfmts.h:5,
                 from init/main.c:18:
./include/asm-generic/siginfo.h:24:20: note: previous definition of 'copy_siginfo' was here
 static inline void copy_siginfo(struct siginfo *to, struct siginfo *from)
                    ^~~~~~~~~~~~
scripts/Makefile.build:310: recipe for target 'init/main.o' failed
make[1]: *** [init/main.o] Error 1
Makefile:1012: recipe for target 'init' failed
make: *** [init] Error 2

It seemed to compile everything when using the default configuration settings, but for some reason I get these errors otherwise. Do you know what I might be doing wrong? If you need more information, just let me know. Thanks!

jleidel, 2017-12-28:
I assume you've built the riscv64-unknown-linux-gnu-* tools to build the kernel.  If so, you need to add "CROSS_COMPILE=riscv64-unknown-linux-gnu-" to your make string as follows: 

$> make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-

...note the trailing "-" on the CROSS_COMPILE argument.

pierceja, 2018-01-04:
Sorry for the late reply. I had to step away from this for a bit during the holidays. Thanks for the response. I tried the command with the cross compile argument, but doing this leads to errors similar to the one shown below:

./include/linux/signal.h:14:20: error: redefinition of 'copy_siginfo'
 static inline void copy_siginfo(struct siginfo *to, struct siginfo *from)
                    ^~~~~~~~~~~~
In file included from ./arch/riscv/include/uapi/asm/siginfo.h:22:0,
                 from ./include/uapi/linux/signal.h:6,
                 from ./include/linux/signal_types.h:10,
                 from ./include/linux/sched.h:27,
                 from ./include/linux/ptrace.h:6,
                 from ./include/linux/ftrace.h:14,
                 from kernel/panic.c:19:
./include/asm-generic/siginfo.h:24:20: note: previous definition of 'copy_siginfo' was here
 static inline void copy_siginfo(struct siginfo *to, struct siginfo *from)

I was looking it up and it seems like errors like this can be caused by multiple inclusions of the same header file. I'm not quite sure what I did wrong to cause this. All of the errors I got refer to this copy_siginfo function by the way. I'm new to all of this so I apologize if this is a trivial error.

palmer-dabbelt, 2018-01-04:
There's some scripts in http://github.com/sifive/freedom-u-sdk that will build a toolchain and vmlinux, those might be helpful here.

pierceja, 2018-01-04:
Thanks! I'll look into those.

stephen422, 2018-01-19:
Can confirm the exact same error appears on my environment (redefinition of 'copy_siginfo').
I suspect this is because the current master branch is based on a newer version of the kernel?
The commit log shows that the linux upstream changes up to 4.15-rc8 are pulled, but the README still points to the 3.14.33 kernel.