riscv-software-src/riscv-tools #164: There is no output when executing "spike bbl vmlinux"

thomascp, 2018-01-03:
I compiled linux and gcc according to README.md in https://github.com/riscv/riscv-tools.
But when executing "spike bbl vmlinux", there is no output of linux booting information.
I tried to change the printk level, but it did not help.
And I input "ctrl+c , then r 100", find out that the simulator is running at the do_idle function.
So, I think maybe it is working fine now, but I dont know how to get the booting informations.
Do I make this question clear? Hope to get your help!
Thanks!

l919898756, 2018-01-24:
I have the same problem

DHSBCGX, 2018-01-24:
I have the same problem

l919898756, 2018-01-24:
@palmer-dabbelt   hello,we have this problem in many days,can you help us.Thanks!

jim-wilson, 2018-01-24:
Try using the sifive/freedom-u-sdk project instead, which is much simpler to build.  It is just "make all" to build everything, and then "make sim" to boot the kernel on the simulator.

sunnystormy, 2018-01-25:
@jim-wilson Just opened a ticket b/c they haven't updated their Linux headers to the new version, which causes the build to exit. If people are still able to get it to work, let me know.

thomascp, 2018-02-02:
@jim-wilson 
Hi jim,
Thanks for your help! It works fine now.
Do you know how to debug the kernel with GUI?
BR,
Peng

DHSBCGX, 2018-02-02:
How does coremark do the migration of riscv?

l919898756, 2018-02-02:
@jim-wilson Hello, do you run the coremark on riscv ?I tried many times did not run successfully, in addition to a lot of mistakes, have you ever been successful? 
Do you have any information I can refer to? thank you very much.

jim-wilson, 2018-02-02:
I'm a binutils/gcc guy.  I've only tried to debug a kernel issue once, and I used printk because kgdb wasn't available.

Running coremark on qemu should be easy.  I haven't tried it, but I've run lots of other stuff on qemu.  I don't have hardware to play with, so I can't comment on that.  You can find user mode qemu in toolchain/bin/qemu-riscv{32,64} which can run programs compiled by the riscv gcc.  If you static link it should just work.  If you link with shared libraries, you probably need a -L option pointing at the sysroot, which is in toolchain/sysroot.

l919898756, 2018-02-04:
@thomascp Hello, I'm still not running spike. can you tell me how you succeeded? did you use sifive / freedom-u-sdk? 
Thank you

thomascp, 2018-02-04:
@l919898756 
Hi, yes, I am using sifive/freedom-u-sdk now.

thomascp, 2018-02-04:
@l919898756 
It is nothing to do with the rootfs.tar.
It seems that you do not have the riscv64-unknown-linux-gnu-gcc.
My tools are in the location : freedom-u-sdk/toolchain/bin/riscv64-unknown-linux-gnu-
You can check it.

jim-wilson, 2018-02-04:
If using sifive/freedom-u-sdk, then don't set RISCV.  That apparently doesn't work.

In the Makefile, changing
 toolchain_dest := $(CURDIR)/toolchain
to
  toolchain_dest := $(RISCV)
might be enough to make this work.

l919898756, 2018-02-05:
Before I use make all, do I need to build my own toolschain for gnu ahead of time, or do I have nothing to do with it? 
Make all will help me build gnu toolschain?
[image]
This is my .banshrc .Am I setting it right?
[image]
He always makes the following mistakes
[image]

l919898756, 2018-02-05:
@jim-wilson Is my MAKEFILE set up correctly, and do I need to install gnu's toolchain in advance before I can use make all? 
Or just download freedom-u-sdk. he'll help me build all the tools.
[image]
I press this to run make all to prompt me for the following error, and I can't find a way to correct it
[image]

jim-wilson, 2018-02-05:
Did you do
  git submodule update --init --recursive
after the git clone?  It looks like the riscv-gnu-toolchain/riscv-gcc dir is missing, which gets created by the git submodule command.

l919898756, 2018-02-05:
I was so stupid, downloading updates was incomplete, except for so many errors, just re-updated, and now I've finally finished compiling and installing, but how can I start the emulator? 
It seems that I can't use spike pk bbl, and there is no RADEME.Is there any document that tells me how to use it? please, thank you very much. @thomascp @jim-wilson

jim-wilson, 2018-02-05:
"make sim" will build a kernel and boot it on spike.  "make qemu" will build a kernel and boot it on qemu.  qemu has more features than spike, so you will be able to do a little more with make qemu than make sim.

l919898756, 2018-02-05:
@jim-wilson hi,It's runing,Thanks very much.But it seem's that I need a username and password,how can I get it?
[image]
Do you know so much, are you slowly studying it? 
Well, you are really very strong. Do you have any blog or articles that I can study? I'm really sorry to ask you questions all the time.

DHSBCGX, 2018-02-05:
What is the system username and password that is started after the compilation is good?

DHSBCGX, 2018-02-05:
I have the same problem,how to solve the question.

jim-wilson, 2018-02-05:
username is root, password is sifive.

jim-wilson, 2018-02-05:
@DHSBCGX  i was wrong about the RISCV environment variable.  The intent is that you either don't set it, or you set it to point to a toolchain install tree that was built via some other means.  So you should not be setting it.  The change I suggested earlier is wrong and I had to rewrite my patch.  This is being discussed here
    https://github.com/sifive/freedom-u-sdk/pull/31

I have a few decades of experience with software development, and as a SiFive employee, I have had to spend a lot of time looking at these software trees.  That is why I know how to answer these kinds of questions.

I don't have a blog, but Palmer Dabbelt has one on the SiFive web site that discusses a number of RISC-V toolchain issues.
    https://www.sifive.com/blog/2017/08/07/all-aboard-part-0-introduction/

jim-wilson, 2018-02-06:
Please start new issues for new questions.

There are Debian, Fedora, Gentoo, and FreeBSD ports in progress.  And probably others too.  If you want a native compiler, it is probably simpler to use one of those than to use freedom-u-sdk.  I've been using the fedora port, second bootstrap image.  They now have a third bootstrap image that I haven't tried yet.
https://fedoraproject.org/wiki/Architectures/RISC-V

priyanka611, 2018-11-28:
i compiled linux 4.19 for riscv as given in readme n when executing 'spike bbl vmlinux' getting stucked between as shown 
please give me suggestion for this issue

priyanka@priyanka-Veriton-M6650G:~/Desktop/RISCV/riscv-tools/riscv-pk/build$ spike bbl vmlinux
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.19.0-29681-g998246b (priyanka@priyanka-Veriton-M6650G) (gcc version 7.2.0 (GCC)) #2 SMP Wed Nov 28 18:06:45 IST 2018
[    0.000000] bootconsole [early0] enabled
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000fffffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000ffffffff]
[    0.000000] software IO TLB: mapped [mem 0xfa3fe000-0xfe3fe000] (64MB)
[    0.000000] elf_hwcap is 0x112d
[    0.000000] percpu: Embedded 17 pages/cpu @(____ptrval____) s29912 r8192 d31528 u69632
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516615
[    0.000000] Kernel command line: 
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 1988740K/2095104K available (5453K kernel code, 333K rwdata, 1745K rodata, 225K init, 807K bss, 106364K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x24e6a1710, max_idle_ns: 440795202120 ns
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [tty0] enabled
[    0.000000] bootconsole [early0] disabled

jim-wilson, 2018-11-28:
New issues for new questions please.  This isn't related to the previous discussion.

The default serial driver is the one that works best on the HiFive Unleashed board. It unfortunately does not work in spike. To get spike working, you need to enable
CONFIG_HVC_RISCV_SBI=y
in conf/linux_defconfig and rebuild.