riscv-software-src/riscv-tools #165: not able to create :riscv32-unknown-elf-gcc

angrej2504, 2018-01-07:
Hi,

I am using following doc to build tool-chain 
https://riscv.org/wp-content/uploads/2015/02/riscv-software-stack-tutorial-hpca2015.pdf
at slide 18, it is mentioned to Generate RV32 code.
Trying following commands:
$ cd ${SRCDIR}/riscv-tools/riscv-gnu-toolchain
$ ./configure --prefix=$RISCV
$ make XLEN=32 linux

but getting following error 
[image]
could you please help me out.

Angrej Singh

jim-wilson, 2018-08-08:
Instructions from 2015 aren't useful.  RISC-V is new, and there have been major changes to everything.  Also, riscv-tools is old, obsolete, unsupported and should not be used.  If you want up to date compiler sources, you have to git clone riscv-gnu-toolchain directly.  riscv-gnu-toolchain has a README.md file with instructions on how to build it.

I think the only way you can get that error is if you already built a toolchain, and are trying to build a second one without removing the first one.  If you have a stamps dir, then you already have done a build here.  You can remove a toolchain with "make clean".  But generally, it is better to create a build dir, cd into the build dir, and then use "../riscv-gnu-toolchain/configure ..." to configure.  That way, you can have multiple build trees, and can easily remove a build tree with rm -rf.

mahsoommoosa42, 2019-02-08:
Hey, jim. I'm trying out the swerv-ISS. They require riscv32-unknown-build compiled with newlib. How do you do this?

jim-wilson, 2019-02-08:
First of all, don't use riscvriscv-tools.  It is broken.  Use riscv/riscv-gnu-toolchain instead.  It has a README file that explains how to build it.

Or you can try grabbing pre-built toolchains.  SiFive has some available at sifive.com/boards.

You can get 32-bit code from a 64-bit multilib toolchain.  Just add -march=rv32imac -mabi=ilp32 or whatever exact arch/abi you need.  We don't copies of the libraries built for every possible arch/abi combination, but we do have most of the common ones.  The SiFive toolchains are 64-bit multilib.