riscv-software-src/riscv-tools #174: riscv32-unknown-elf-gcc: error trying to exec 'cc1': execvp: No such file or directory

sriteja777, 2018-02-17:
./build.sh completed successfully 
but when executing the below command 

`riscv32-unknown-elf-gcc -o hello hello.c`

I am getting the following error

`riscv32-unknown-elf-gcc: error trying to exec 'cc1': execvp: No such file or directory`

but when I tried with the command 

riscv64-unknown-elf-gcc -o hello hello.c

it worked out properly.
for my lab assignments I  should use 32 bit riscv

Please provide a solution for it
[image]

jim-wilson, 2018-02-22:
You can get 32-bit code from a 64-bit compiler, just add -march=rv32gc -mabi=ilp32d for instance, or other options if you want other arch/abi combination.

Since you have a working 64-bit compiler, you might try looking at the 32-bit and 64-bit build and install trees to see what is different.  Besides a few dirs/files being renamed, they should otherwise be the same.

If you have a build log, try looking at that for errors.  Probably something went wrong during your build.

riscv-tools is not well maintained.  It is easier to build toolchains from riscv/riscv-gnu-toolchain.  if you need a linux kernel, it is easier to build sifive/freedom-u-sdk.

sriteja777, 2018-02-22:
But after I get the 32 code from 64 bit compiler, How do I execute it? 
When I run 

 `spike run a.out`
 
 it is is not terminating, even I tried using 
 
`spike --isa=rv32 pk a.out`
 
even then it is not terminating 
How do I execute it after compiling successfully

jim-wilson, 2018-02-22:
Try qemu.  It is generally more useful than spike.  You will need a 32-bit qemu, and for user programs you will want to use a user mode qemu not a system qemu.

jim-wilson, 2018-02-22:
In riscv-gnu-toolchain, you can build qemu with "make stamps/build-qemu".  In freedom-u-sdk, you can do "make qemu" to build qemu and boot linux on it.

sriteja777, 2018-02-23:
yaa, ok thanks