riscv-software-src/riscv-tools #176: No child processes

sarisercan, 2018-02-21:
```
/tmp/ccbuxf4K.o: In function `_start':
(.text.init+0x40): undefined reference to `tohost'
/tmp/ccbuxf4K.o: In function `.L0 ':
(.text.init+0x108): undefined reference to `tdat'
/tmp/ccbuxf4K.o: In function `.L18':
(.text.init+0x138): undefined reference to `tdat'
(.text.init+0x134): undefined reference to `fail'
/tmp/ccbuxf4K.o: In function `test_2':
(.text.init+0x164): undefined reference to `fail'
collect2: error: ld returned 1 exit status
make[1]: *** [rv64ui-p-ld_sd_lock_unlock_func] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** wait: No child processes.  Stop.
make: *** [isa] Error 2
make: *** Waiting for unfinished jobs....
```

We are getting this error when we are trying to compile rocket-chip. We are getting this error when we are trying to compile rocket-chip. We have tried to change the test cases with simple ld.lock and sd.lock instructions that we added the instruction set. However, we had this error. We have focused the test.ld, syscall.c files but could't figure out the issue. Is there any solution to this?

jim-wilson, 2018-02-21:
If you want to add your own testcase to riscv-tests and have it work, then you must exactly copy the form of the existing tests.  The missing fail symbol indicates that you are missing a call to TEST_PASSFAIL in your testcase.  The missing tohost indicates a missing call to RVTEST_DATA_BEGIN.  The tdat has me puzzled, but that is perhaps a bug in some macro that you added.  See for instance isa/rv64ui/add.S.

sarisercan, 2018-02-22:
We added exactly copy of the existing tests. Also, we added new macos. Right now, we got rid of previous errors but we have "failure to make vcd file" error.

**Tests we are trying to execute:**

Test is executed without error while only testing this case.
`TEST_LD_SEC_OP( 2, 0x00ff00ff00ff00ff, tdat, ld.lock x30, 0(x1) );`

Test is also executed without error while only testing this two cases.
```
TEST_LD_SEC_OP( 2, 0x00ff00ff00ff00ff, tdat, ld.lock x30, 0(x1) );
TEST_LD_SEC_OP( 3, 0xff00ff00ff00ff00, tdat, ld.lock x30, 8(x1) );
```

Also, test is executed without error while only testing this case.
`TEST_LD_SEC_OP( 4, 0x0ff00ff00ff00ff0, tdat, ld.lock x30, 16(x1) );`

However, when we tried to test with these three cases together, we had error below.
```

TEST_LD_SEC_OP( 2, 0x00ff00ff00ff00ff, tdat, ld.lock x30, 0(x1) );
TEST_LD_SEC_OP( 3, 0xff00ff00ff00ff00, tdat, ld.lock x30, 8(x1) );
TEST_LD_SEC_OP( 4, 0x0ff00ff00ff00ff0, tdat, ld.lock x30, 16(x1) );
```

```
[Installing project riscv-tests

RISC-V Toolchain installation completed!
./emulator-freechips.rocketchip.system-DefaultConfig-debug +max-cycles=100000000 +verbose -voutput/rv64ui-p-ld_sd_lock_unlock_func.vcd output/rv64ui-p-ld_sd_lock_unlock_func 3>&1 1>&2 2>&3 | /home/sercan/secchisel_v2/src/rocket-chip/riscv/bin/spike-dasm  > output/rv64ui-p-ld_sd_lock_unlock_func.out && [ $PIPESTATUS -eq 0 ]
Makefile:43: recipe for target 'output/rv64ui-p-ld_sd_lock_unlock_func.vcd' failed
make: *** [output/rv64ui-p-ld_sd_lock_unlock_func.vcd] Error 1](url)
```

jim-wilson, 2018-02-22:
I don't know anything about the chisel stuff.  You might try running commands manually and looking at output and return values to see which part of the pipe commands failed.

sarisercan, 2018-02-23:
Thanks a lot @jim-wilson. 
How do I run commands manually? Since I'm a beginner I couldn't file any documentation about manual running.

jim-wilson, 2018-02-23:
I meant look at the commands echoed to the build log, and try typing them in manually to see what they do.