riscv-software-src/riscv-tools #184: "riscv64-unknown-elf-gcc command not found"

tyskink, 2018-03-17:
Hello, I installed the toolchain in Ubuntu. It compiled one hello file successfully. but while I restarted the Ubuntu and compile it again, it said "riscv64-unknown-elf-gcc command not found"
even I tried build the toolchain again, it still not work. Has anybody met this issue before?
[image]

jim-wilson, 2018-03-17:
Maybe you need to put the toolchain install dir on your path?  Or try typing the full path to the toolchain instead of just the toolchain name?

seldridge, 2018-03-17:
I think you just need to add `$RISCV/bin` to your `$PATH` again. There should be some instructions in the README about this:

```bash
export PATH=$RISCV/bin:$PATH
```

sharinoz, 2018-05-13:
hi, I have the same problem here. I tried to fix the $PATH variable, but it still doesn't work

jim-wilson, 2018-08-25:
The problem is on your end.  You need to do some basic debugging.

What did you set $RISCV to when you built the toolchain?  Is that environment variable still set to the same value?  Is the $RISCV install directory still there?  Is $RISCV/bin on your path?  Do you have a $RISCV/bin/riscv64-unknown-elf-gcc file?  Do you have any files in $RISCV/bin? What are they?  If you have riscv64-unknown-elf-gcc, then what does  "ldd riscv64-unknown-elf-gcc" say?

sharinoz, 2018-08-27:
some how this line is working for me now, thanks anyway

RubelAhmed57, 2018-11-13:
1. compiled it using gcc
2. ~$ spike pk hello
got the following
[image]

I dont want the isa simulator, rather I want rocket core will execute the instructions. How to do that?

jim-wilson, 2018-11-13:
The problem you are reporting doesn't match the title.  Please open new issues for new problems.  You only showed part of the error message, so I can't tell what went wrong.  I don't know how to simulate rocket, that isn't a tools question.  Try asking on one of the rocket related forums.  See for instance https://riscv.org/risc-v-cores/ which has several rocket related links.

saranrajsridhar, 2019-07-08:
hello guys 
i have same issues i have tried hello.c after installing it was fine 
but after restart its not working 
but i have applied that path as you said still i get same error

jim-wilson, 2019-07-08:
Check that the program files are actually in the install dir where you think they are, and check that your PATH includes the install dir where the programs were installed.  Also check for linux versus elf compiler confusion, e.g. if you built and installed a linux compiler then you won't be able to run an elf compiler until you build one.  If "which riscv64-unknown-elf-gcc" reports nothing then either your PATH environment variable is wrong or you don't have a riscv64-unknown-elf-gcc compiler installed.

saranrajsridhar, 2019-07-08:
But I got the output once ,okay I wil try to reinstall the risc v again

saranrajsridhar, 2019-07-09:
sir,could you help me from first step to install it
i have no idea about the path where $RISCV/bin
[image]
[image]
[image]

bagyavinoth, 2019-09-17:
> hich riscv64-unknown-elf-gcc

Even in my case i get below error "make[1]: riscv64-unknown-elf-gcc: Command don’t found".
And "which riscv64-unknown-elf-gcc" reports nothing.
Then i understood when i was building the [toolchain](https://github.com/riscv/riscv-gnu-toolchain), i gave "make linux" which means it builds a linux compiler.
Just "make" commands builds an elf compiler & the command "which riscv64-unknown-elf-gcc" would list the toolchain path.
This would solve the problem.

jim-wilson, 2019-09-17:
If you use --enable-linux when configuring, then you can just type make to build a linux toolchain.

bhawandeepsingh, 2021-02-13:
@jim-wilson  I am facing the same issue, and have tried things told in various threads, but nothing has helped.
I am working on Ubuntu, I already have riscv-linux-gnu-gcc installed. But when I try to compile a test I get the above error.
I used 
./configure --prefix=$RISCV/target --enable-linux

Any idea what I can try ? Please let me know if more information will be helpful, I will be glad to provide, thanks.

jim-wilson, 2021-02-13:
--enable-linux is for riscv-gnu-toolchain, not riscv-tools.  It won't do anything useful for riscv-tools.

You mention having riscv-linux-gnu-gcc installed, and then getting an error for a missing riscv64-unknown-elf-gcc.  Those are two different compilers.  You need the riscv64-unknown-elf-gcc to build riscv-tools.

This bug report is getting very confused.  riscv-tools and riscv-gnu-toolchain are different packages.

bhawandeepsingh, 2021-02-13:
Thanks @jim-wilson 
So, now, I have the riscv-gnu-toolchain installed on my VM.
I do not have riscv64-unknown-elf-gcc. and do not have the admin rights or the disk space on VM to install it.
My workload is by default looking for riscv64-unknown-elf-gcc
The workload to compile is - 
https://github.com/riscv/riscv-tests

Can I make it use riscv-gnu-toolchain  ? 
I tired - 
make CC=riscv64-linux-gnu-gcc CXX=riscv64-linux-gnu-g++

But, I still got the same error.
Then I added -static to my build commands in the make file and now I am getting - 
make: invalid option -- 'a'
make: invalid option -- 'c'

Thanks a lot once again for your timely response, I will be thankful if you can show me way out of this, thanks.

jim-wilson, 2021-02-14:
riscv-tests are ISA tests that assume a bare metal target.  I don't know if a linux compiler can work for them.  They are designed to work with an embedded elf toolchain.

You can build an embedded elf compiler from riscv-gnu-toolchain.  See the README.md file.  Or you can install one.  There are several on the SiFive web site in the software section, including the SiFive toolchain and some third party toolchain builds.  You don't need admin rights to install one in your home dir, but you do need enough disk space.

If you built an embedded elf toolchain from riscv-gnu-toolchain and it isn't working, then try doing some basic consistency checks.  Does the install dir exist?  Is the compiler in the install dir?  If the install dir on your path?  Does the program in the install dir have the same name as the program that the riscv-tools makefile is trying to run.  If you cd into the install bin dir and try to run the program does it work?

I'd guess that you used "-static" instead of "--static" and/or that you added it in the wrong place.  POSIX says that -s -t -a -t -i -c is equivalent to -static, but POSIX doesn't say anything about options that start with two dashes, so the GNU toolchain uses long options like --static.  However, not all programs obey the POSIX rules for command line options, and not everyone modifying the GNU toolchain remembers the rules, so there are some programs and some options that break the rule.  But in general if you try -static and you get an error that says -a is invalid, then try --static instead.

NervenCid, 2021-04-12:
Hello to all there are any solution i tried the documentation here:

https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html

and ran these commands:

`cd linux
git checkout v5.4.0
make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- defconfig
`

But i got these errors:

`
*** Default configuration is based on 'defconfig'
scripts/Kconfig.include:35: compiler 'riscv64-unknown-linux-gnu-gcc' not found
make[1]: *** [scripts/kconfig/Makefile:83: defconfig] Error 1
make: *** [Makefile:567: defconfig] Error 2
`

Thank you

jim-wilson, 2021-04-12:
This was also asked in #342, so ignoring this one.

jim-wilson, 2022-02-01:
It is better to create new issues than to add to old ones.  Your problem is probably different from the others here.

What exact command is giving you the "riscv64-unknown-elf-gcc not found" error?  And what is the exact error?  You probably got more than one line of error output.  The context of the error could be important.

If you are getting the error while building riscv-tools, then riscv-tools creates build.log files in each subdir that it builds.  So try looking at those build.log files to see which one has the error, and this might give a clue as to what is wrong.

yagyamundra, 2022-02-02:
@jim-wilson I am trying to run the command "make hello" to run Hello World program to test riscv-gnu-toolchain. This is the whole thing I am getting:

Running hello world sim
make[1]: Entering directory '/home/yagya/Desktop/chromite/benchmarks'
Compiling Hello
riscv64-unknown-elf-gcc -I./common -I./hello -DCONFIG_RISCV64=True \
			-DCUSTOM=True -mcmodel=medany -static -std=gnu99 -O2 -ffast-math \
			-fno-common -fno-builtin-printf -march=rv64imfd -mabi=lp64d -w -static \
			-nostartfiles -lgcc -T ./common/link.ld -o output/hello.riscv ./hello/hello.c \
			./common/syscalls.c ./common/crt.S
make[1]: riscv64-unknown-elf-gcc: Command not found
make[1]: *** [Makefile:45: hello] Error 127
make[1]: Leaving directory '/home/yagya/Desktop/chromite/benchmarks'
make: *** [Makefile:136: hello] Error 2

Thanks a lot once again for your timely response.

yagyamundra, 2022-02-03:
@jim-wilson I have added compiler path to the directory and still which riscv64-uknown-elf-gcc was not showing anything. I tried sudo apt-get install gcc-riscv64-unknown-elf and now I am getting a different error instead :

./hello/hello.c:1:10: fatal error: stdio.h: No such file or directory
    1 | #include <stdio.h>
      |          ^~~~~~~~~
compilation terminated.
In file included from ./common/syscalls.c:3:
/usr/lib/gcc/riscv64-unknown-elf/9.3.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:45: hello] Error 1

I dont understand this issue. I have all the files installed in my directory /opt/riscv and even after giving that directory as my PATH, it is taking up the values from the directory /usr/lib

Please help with this issue

jim-wilson, 2022-02-03:
What OS do you have?  Most linux distros won't have a package containing a riscv64-unknown-elf compiler.  Debian added one.  Ubuntu takes packages from Debian, but due to bad timing, Ubuntu 20.04 LTS has broken support for riscv64-unknown-elf.  It has the compiler, but not the library.  This is fixed in later Ubuntu versions.  Not sure exactly when, but is OK with the latest version 21.10.    So if you have Ubuntu 20.04, then installing a package won't work.  You need an OS upgrade, or you build it yourself, or you download a pre-built one from somewhere else.

You need to put /opt/riscv/bin on your path, not /opt/riscv.  Verify that you have binaries in /opt/riscv/bin, and verify that they work, e.g. try running /opt/riscv/bin/riscv64-unknown-elf-gcc directly.   There might be something wrong there.  For instance, if you built and installed as root, maybe they are only executable by root, and you need chmod/chown to fix them so you can use them.