riscv-software-src/riscv-tools #187: failure on linux compilation

allencho1222, 2018-04-04:
I followed the lowRISC guide to compile linux for untethered version:

# set up the RISCV environment variables
cd $TOP/riscv-tools
curl https://www.kernel.org/pub/linux/kernel/v3.x/linux-3.14.41.tar.xz \
  | tar -xJ
cd linux-3.14.41
git init
git remote add origin https://github.com/lowrisc/riscv-linux.git
git fetch
git checkout -f -t origin/untether-v0.2
make ARCH=riscv defconfig
make ARCH=riscv -j vmlinux

And I got following errors : (I enter the 'sudo make ARCH=riscv -j8 vmlinux')

make: riscv64-unknown-linux-gnu-gcc: Command not found
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CC      scripts/mod/empty.o
/bin/sh: 1: riscv64-unknown-linux-gnu-gcc: not found
scripts/Makefile.build:308: recipe for target 'scripts/mod/empty.o' failed
make[2]: *** [scripts/mod/empty.o] Error 127
make[2]: *** Waiting for unfinished jobs....
  CC      scripts/mod/devicetable-offsets.s
/bin/sh: 1: riscv64-unknown-linux-gnu-gcc: not found
scripts/Makefile.build:204: recipe for target 'scripts/mod/devicetable-offsets.s' failed
make[2]: *** [scripts/mod/devicetable-offsets.s] Error 127
scripts/Makefile.build:455: recipe for target 'scripts/mod' failed
make[1]: *** [scripts/mod] Error 2
Makefile:527: recipe for target 'scripts' failed
make: *** [scripts] Error 2

The error message says that it can't find 'riscv64-unknown-linux-gnu-gcc'. However, I already installed the riscv tool chain including riscv64-unknown-linux-gnu-gcc, even for newlib version.
In addition, I added the path of 'riscv64-unknown-linux-gnu-gcc into PATH env variable, and 'which' command shows where it is.

Furthermore, I tried same thing by using other commands such as including CROSS_COMPILE=riscv64-unknown-linux-gnu-'. But, all fails.

seldridge, 2018-04-04:
> make: riscv64-unknown-linux-gnu-gcc: Command not found

You do not have `riscv64-unknown-linux-gnu-gcc` on your path. See the instructions for building a Linux toolchain here: https://github.com/riscv/riscv-gnu-toolchain#installation-linux. Note: after you build that, a `make install` will install that in `$RISCV/bin` (assuming that you set up your `$RISCV` environment variable). You will then have both a newlib (`*-unknown-elf-*`) and linux (`*-unknown-linux-gnu-*`) toolchain available. The latter is needed to build the Linux kernel.

Also, no need to use `sudo` for this. That will only complicate things as it won't find things on your user `$PATH`...

allencho1222, 2018-04-04:
@seldridge Thks for replying. But as you can see from pictures, there's no problem on PATH variable.
[image]
[image]

seldridge, 2018-04-04:
Your user `$PATH` will not have the same path as the root user. When you run `sudo`, you're running in a root shell and will have the `$PATH` of root. As an example, running `sh -c 'echo $PATH'` as me and with `sudo` will produce two different paths. My user path has my local install of the toolchain in `/opt/riscv/bin` while the root path does not:
```
> sh -c 'echo $PATH'
/home/se/usr/bin:/home/se/usr/stow/anaconda3/bin/:/home/se/perl5/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/opt/riscv/bin:/usr/local/texlive/2016/bin/x86_64-linux:
> sudo sh -c 'echo $PATH'
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
```
I think that you can build the kernel without `sudo`. Can you try a `make ARCH=riscv -j vmlinux` without `sudo`?

Note: running `sudo echo $PATH` will show your user path as bash will expand `$PATH` _before_ the command is executed.

seldridge, 2018-04-04:
You may need to explicitly specify the build target as `vmlinux` per the lowRISC guide. I haven't done this in a while, but can test locally (at some point).

allencho1222, 2018-04-04:
@seldridge running without sudo command shows me a problem on 'Permission Denied':

[image]

jim-wilson, 2018-04-04:
The fact that earlier builds were built with sudo means that your build tree has files owned by root, and you will not be able to overwrite them without sudo.  You should not need sudo to do a build.  I'd suggest deleting your current linux build tree very carefully using sudo rm -rf, and starting a new one without sudo.

allencho1222, 2018-04-04:
It seems that the problem was 'sudo' command like what @jim-wilson says. However, after make new copy of linux, I met following new errors:
[image]

jim-wilson, 2018-04-04:
Looks like you might be accidentally building an x86 kernel.  Try looking at the .config file in the linux build tree.  It should have a CONFIG_RISCV=y line near the top.  If you see any other architecture, x86 in particular, then it wasn't configured correctly.

allencho1222, 2018-04-05:
@jim-wilson CONFIG_RISCV is correctly set to 'y'. But, I still have problems like above.

jim-wilson, 2018-04-05:
You should check to see if you have any environment variables set that could affect the compiler.  Maybe you have a global definition of CFLAGS that adds -m64 for instance.

palmer-dabbelt, 2018-10-17:
I'm closing this as it's not a riscv-tools issue.

closed 2018-10-17 by palmer-dabbelt