riscv-software-src/riscv-tools #191: Where to add "-msoft-float" to build pk which doesn't include fpu?

clx7913, 2018-04-17:
I disabled the fpu when I configed the rocket-chip.

When I typed "./fesvr-zynq pk hello",it told me:
`../machine/minit.c:59: FPU not found; recompile pk with -msoft-float`.

But I don't know how to recompile pk with `-msoft-float`,could someone tell me where to add "-msoft-float"?

jim-wilson, 2018-04-17:
When you configure the toolchain, you can specify --with-arch=X and --with-abi=Y options to set the compiler defaults.  That is probably one of the easiest ways to fix this.  Depending on what actual features are in your target, you might want something like --with-arch=rv32imac --with-abi=ilp32 when configuring the toolchain.

The RISC-V gcc port does not actually accept the -msoft-float option, so that isn't a useful comment.  I don't see code to print this error in pk, but maybe you are using an older version of pk.

clx7913, 2018-04-17:
Yes,it's an old version.`https://github.com/riscv/riscv-tools/tree/6136df9cba56afe12c2fa321ef497090917c15aa`

I use this version because `https://github.com/ucb-bar/fpga-zynq` references this version of riscv-tools.

clx7913, 2018-04-17:
I haven't encountered the problem when I use the default config `ZynqConfig`.

But when I implemented dual-core rocket,I encountered this problem.

jim-wilson, 2018-04-17:
I don't know what ZynqConfig is.  I'm a software guy.  I haven't done anything with FPGAs.  All I can suggest is that you build your own toolchain, configured for the appropriate arch and abi.

clx7913, 2018-04-18:
Thank you all the same,I'll try it.

closed 2018-05-01 by clx7913