riscv-software-src/riscv-tools #202: build.sh fails while configuring project riscv-openocd

sharinoz, 2018-05-06:
hi,
when i try to run ./build.sh everytime i get this message:

Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
configure: error: cannot find install-sh, install.sh, or shtool in ".." "../.." "../../.."

i have already installed libtool, shtool and autogen, but still unsuccessful.
what do I need to do in order to complete the build process?

thanks
Shahar

jim-wilson, 2018-05-06:
riscv-tools is unfortunately poorly maintained, and hard to build.  It isn't a good package to try if you are just starting to do risc-v work.  If you just need a toolchain, then try riscv-gnu-toolchain instead.

When you build riscv-openocd, the first thing that happens is that the build scripts run autoreconf, and this creates the riscv-openocd/install-sh file.  If you do not have this file, then something went wrong when autoreconf was run.  You could try something like "git clean -d -x" inside the riscv-openocd dir to get back to the original state, and this should force autoreconf to run again.  If that doesn't work, then there might be some version dependency.  We use Ubuntu 16.04 here.  If you are using Ubuntu 18.04 for instance, then we might need riscv-tools patches to make that work.

sharinoz, 2018-05-08:
just tried it, didn't worked
I guess I will try the riscv-gnu-toolchain
thanks

sharinoz, 2018-05-13:
OK, I tried the gnu-toolchain, it seemed that everything went fine, but then when I try to do ./build.sh I get the same errors!
I tried to check the toolchain, as they explain in here:
https://github.com/riscv/riscv-tools/blob/priv-1.10/README.md#-testing-your-toolchain

and when I try the riscv64-unknown-elf gcc command I get this message:
riscv64-unknown-elf-gcc: command not found

I tried google it, but didn't found any solution that worked for me
can you help?

sharinoz, 2018-05-15:
the command "which riscv64-unknown-elf-gcc" gives no answer at all

in the bin dir , ls says:

riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-gcc         riscv64-unknown-linux-gnu-gdb       riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-ar         riscv64-unknown-linux-gnu-gcc-7.2.0   riscv64-unknown-linux-gnu-gfortran  riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-as         riscv64-unknown-linux-gnu-gcc-ar      riscv64-unknown-linux-gnu-gprof     riscv64-unknown-linux-gnu-run
riscv64-unknown-linux-gnu-c++        riscv64-unknown-linux-gnu-gcc-nm      riscv64-unknown-linux-gnu-ld        riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-c++filt    riscv64-unknown-linux-gnu-gcc-ranlib  riscv64-unknown-linux-gnu-ld.bfd    riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-cpp        riscv64-unknown-linux-gnu-gcov        riscv64-unknown-linux-gnu-nm        riscv64-unknown-linux-gnu-strip
riscv64-unknown-linux-gnu-elfedit    riscv64-unknown-linux-gnu-gcov-dump   riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-g++        riscv64-unknown-linux-gnu-gcov-tool   riscv64-unknown-linux-gnu-objdump
rotemshahar@asic2-serv3:~/rocket/riscv-gnu-toolchain/bin$

jim-wilson, 2018-05-15:
You built a linux compiler not an embedded elf compiler.  Which one you want depends on what you are doing.  If you want the other one, then build the other one.

firstmover, 2019-05-10:
I removed the whole git repo and installed it again. Problem solved.