riscv-software-src/riscv-tools #209: config.status:error cannot find input file:'Makefile.in'

XRNGX, 2018-05-22:
linux version is rhel6.7

XRNGX, 2018-05-22:
$git submodule update --init --recursive
$ export RISCV=/path/to/install/riscv/toolchain
$sudo dnf install autoconf automake @development-tools curl dtc libmpc-devel mpfr-devel gmp-devel libusb-devel gawk gcc-c++ bison flex texinfo gperf libtool patchutils bc zlib-devel expat-devel
then run ./build.sh and get this message
config.status:error cannot find input file:'Makefile.in'

XRNGX, 2018-05-23:
thanks,and this is the build.log

checking for realpath... yes
checking whether to build a release... no
checking whether to build Doxygen as HTML... yes
checking whether to build Doxygen as PDF... no
checking whether to enable verbose JTAG I/O messages... no
checking whether to enable verbose USB I/O messages... no
checking whether to enable verbose USB communication messages... no
checking whether to enable malloc free space logging... no
checking whether to enable ZY1000 minidriver... no
checking whether to enable dummy minidriver... no
checking whether standard drivers can be built... yes
checking for LIBUSB1... yes
configure: libusb-1.0 header bug workaround: LIBUSB1_CFLAGS changed to "-isystem /usr/include/libusb-1.0"
checking for LIBUSB0... yes
checking for HIDAPI... no
checking for HIDAPI... no
checking for HIDAPI... no
checking for LIBFTDI... no
checking for LIBFTDI... no
checking for LIBJAYLINK... no
checking for environ in unistd.h and stdlib.h... yes
configure: creating ./config.status

jim-wilson, 2018-05-23:
The very next thing that should be in the build.log file is 
```
config.status: creating Makefile
```
so the config.status script failed for some reason.  

It isn't clear what is wrong.  This could be a problem with your shell.  I'm using /bin/bash on a Ubuntu 16.04LTS system.

Or maybe the config.status file is corrupted.  You could try deleting it and rerunning the build to generate it again.

Or maybe your riscv-openocd dir got corrupted somehow.  Try running "git status" in that dir to see if anything is missing or modified.

The config.status code in question starts with
```
eval set X "  :F $CONFIG_FILES  :H $CONFIG_HEADERS    :C $CONFIG_COMMANDS"
shift
for ac_tag
...
```
and should end up finding ../Makefile.in.  You could try running config.status manually.  If you do something like "sh -x config.status >& tmp.file", then you should see something like 
```
+ ac_file=Makefile
+ shift
+ case $ac_mode in
+ ac_file_inputs=
+ for ac_f in '"$@"'
+ case $ac_f in
+ test -f Makefile.in
+ case $ac_f in
+ test -f ../Makefile.in
+ ac_f=../Makefile.in
+ case $ac_f in
+ as_fn_append ac_file_inputs ' '\''../Makefile.in'\'''
+ eval 'ac_file_inputs+=$2'
++ ac_file_inputs+=' '\''../Makefile.in'\'''
++ printf '%s\n' Makefile.in
++ sed 's|^[^:]*/||;s|:[^:]*/|, |g'
+ configure_input='Generated from Makefile.in by configure.'
+ test xMakefile '!=' x-
+ configure_input='Makefile.  Generated from Makefile.in by configure.'
+ printf '%s\n' 'config.status:1284: creating Makefile'
```

in the output file.

XRNGX, 2018-05-25:
Thanks, I solved it. The problem is that I use yum install automake, the installed version is 1.11, and the required version is 1.15, but the warning is not written this way.