riscv-software-src/riscv-tools #212: The benchmark of dhrystone score was lower than the Official data

IC-Dream, 2018-05-24:
Hi,

Since you have closed my issues about this question, but i am still confused and  want to consult you. i mentioned that there are two build modes: a generic ELF/Newlib toolchain and a more sophisticated Linux-ELF/glibc toolchain. So the default build modes was the former, right? could we switch to the latter to run simulation(Verilator) and how to switch? And, if it is permitted, could you descriped it in detail, since i am a hardware engineer~

The previous issue link:
https://github.com/riscv/riscv-tools/issues/211

Thanks  a lot for your help

jim-wilson, 2018-05-24:
I don't know anything about verilator, I'm a software engineer.  riscv-tools in general is poorly supported and difficult to build.  I wouldn't recommend use of it.

If you want a toolchain, I'd suggest riscv-gnu-toolchain instead.  If you want to build a linux system, I would suggest sifive/freedom-u-sdk instead, which will build a linux toolchain, kernel, and buildroot.  It also has qemu that you can use to boot linux.

IC-Dream, 2018-07-17:
@seldridge Hi, is there any ways to improve the score of dhrystone benchmarks ? I try to modify the CFLAG of riscv64-*-gcc to compile the dhrystone, but still can't reach the official data(1.72DIMPS/Mhz). 

Thanks