riscv-software-src/riscv-tools #215: Different interpretation of one elf file

pentin-as, 2018-05-29:
I run the same toolchain (not recompiled) with the same elf file (not recompiled) on different PCs and have different results.

For example, I cause:
riscv64-unknown-elf-objdump -S mt-matlul.riscv

On PC1 (1.txt) from address 0x80001048 begins "barrier, and from the address 800011a0 - "matmul".
On PC2 (2.txt) from the address 0x80001048 begins "matmul", and from the address 80001290 - "barrier".

Because of this, I have different times on different PCs to run the same test. How can it be?

[1.txt](https://github.com/riscv/riscv-tools/files/2047393/1.txt)
[2.txt](https://github.com/riscv/riscv-tools/files/2047395/2.txt)
[mt-matmul.zip](https://github.com/riscv/riscv-tools/files/2047398/mt-matmul.zip)

jim-wilson, 2018-05-30:
I don't believe that it is the same elf file.  Simple utilities like objdump and objcopy won't create different output for the same elf file.

It is possible that the same compiler binary and the same source file can produce different elf files on two different computers with slightly different OS installs.  This is the most likely explanation for what you are seeing.  Maybe you rebuilt the program on one computer and didn't realize it?

pentin-as, 2018-05-31:
Thank you.
Indeed, there were different elf files. My mistake.
I'm sorry.

closed 2018-05-31 by pentin-as