riscv-software-src/riscv-tools #219: Is Gcov work on riscv arch?

BaikalHu, 2018-06-13:
Hi all,

Is libgcov.a and '-fprofile-arcs -ftest-coverage' option work on riscv arch?
How to use libgcov.a on SiFive E31 RISC-V Core? I just want to use uart to transfer the GCOV data to PC,could someone give the suggestion? Thanks.

jim-wilson, 2018-06-13:
The gcov support is mostly target independent.  It should work fine for RISC-V.  For an embedded target, the main problem will be getting the results off the board.  Normally gcov will write results to a file on exit.  If you don't have file I/O, then you have to get the results off the board some other way.  Using gdb is one possibility.  Which ever way you choose, you may have to write some code yourself, as I don't think there is any standard solution for this.  If your program doesn't call exit, there are some hooks you can call.  __gcov_reset resets the gcov data.  __gcov_dump collects the data for dumping to a file, or for grabbing some other way.

A web search for "gcov gdb embedded" shows a number of links that talk about using gcov in an embedded environment with gdb.

BaikalHu, 2018-06-14:
@jim-wilson  Thanks for your suggestion. I'll try to do some tests.

closed 2018-06-14 by BaikalHu