riscv-software-src/riscv-tools #221: Difficulty Uploading demo_gpio to E300 Freedom E300

smiqbal, 2018-06-15:
I’m having some trouble with uploading the demo_gpio program to the Arty board.

I run this command make upload PROGRAM=demo_gpio BOARD=freedom-e300-arty:

This is the output:
work/build/openocd/prefix/bin/openocd -f bsp/env/freedom-e300-arty/openocd.cfg &
/home/syediqbal/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix/bin/riscv64-unknown-elf-gdb software/demo_gpio/demo_gpio --batch -ex “set remotetimeout 240” -ex “target extended-remote localhost:3333” -ex “monitor reset halt” -ex “monitor flash protect 0 64 last off” -ex “load” -ex “monitor resume” -ex “monitor shutdown” -ex “quit” &&
echo "Successfully uploaded ‘demo_gpio’ to freedom-e300-arty."
Open On-Chip Debugger 0.10.0+dev (2018-06-11-16:11)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport “jtag”. To override use 'transport select '.
Error: libusb_claim_interface() failed with LIBUSB_ERROR_BUSY
Error: unable to open ftdi device with vid 15ba, pid 002a, description ‘Olimex OpenOCD JTAG ARM-USB-TINY-H’, serial ‘’ at bus location '’

Info : accepting ‘gdb’ connection on tcp/3333
Error: Device ID 0x20000913 is not known as FESPI capable
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use ‘gdb_memory_map disable’.
Error: attempted ‘gdb’ connection rejected
Remote connection closed
"monitor" command not supported by this target.
“monitor” command not supported by this target.
You can’t do that when your target is `exec’
“monitor” command not supported by this target.
“monitor” command not supported by this target.
Successfully uploaded ‘demo_gpio’ to freedom-e300-arty.

I’ve gone through this other issue on the forum and I’ve changed my openocd.cfg file accordingly but this still has not fixed the problem. They suggest killing other openocd process but I can’t find it in the list of active processes on my machine. One relevant detail is that I am building the MCS file from scratch in the freedom repository. Any help is greatly appreciated.

jim-wilson, 2018-06-15:
You are asking a hardware question in a software group.  I'd suggest finding someplace better to ask hardware questions.  Such as the sifive/freedom-e-sdk github project, or an SiFive hardware related forum.

palmer-dabbelt, 2018-06-15:
`LIBUSB_ERROR_BUSY` usually means something else is running attached to that USB port.  There's probably another OpenOCD instance running.

Like Jim said, this isn't really the right place to ask the question.

closed 2018-06-19 by smiqbal