riscv-software-src/riscv-tools #228: Error building binutils  in riscv-gnu-toolchain target with GCC 4.9.4

awxhw, 2018-08-06:
when building binutils, I met the errors as follows,

### Can't locate Locale/Messages.pm in @INC (you may need to install the Locale::Messages module) (@INC contains: /usr/share/texinfo/lib/Unicode-EastAsianWidth/lib /usr/share/texinfo /usr/lib/perl5/site_perl/5.22.1/x86_64-linux /usr/lib/perl5/site_perl/5.22.1 /usr/lib/perl5/5.22.1/x86_64-linux /usr/lib/perl5/5.22.1 .) at /usr/share/texinfo/Texinfo/Report.pm line 49.
BEGIN failed--compilation aborted at /usr/share/texinfo/Texinfo/Report.pm line 49.
Compilation failed in require at /usr/share/texinfo/Texinfo/Parser.pm line 50.
BEGIN failed--compilation aborted at /usr/share/texinfo/Texinfo/Parser.pm line 50.
Compilation failed in require at /usr/bin/makeinfo line 101.
BEGIN failed--compilation aborted at /usr/bin/makeinfo line 101.
make[4]: *** [bfd.info] Error 2
make[4]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd/doc'
Making info in po
make[4]: Entering directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd/po'
( if test 'x/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/bfd/po' != 'x.'; then \
	    posrcprefix='/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/bfd/'; \
	  else \
	    posrcprefix="../"; \
	  fi; \
	  rm -f SRC-POTFILES-t SRC-POTFILES \
	    && (sed -e '/^#/d' \
	            -e '/^[ 	]*$/d' \
		    -e "s@.*@	$posrcprefix& \\\\@" < /home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/bfd/po/SRC-POTFILES.in \
		| sed -e '$s/\\$//') > SRC-POTFILES-t \
	    && chmod a-w SRC-POTFILES-t \
	    && mv SRC-POTFILES-t SRC-POTFILES )
( rm -f BLD-POTFILES-t BLD-POTFILES \
	    && (sed -e '/^#/d' \
	            -e '/^[ 	]*$/d' \
		    -e "s@.*@	../& \\\\@" < /home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/riscv-binutils-gdb/bfd/po/BLD-POTFILES.in \
		| sed -e '$s/\\$//') > BLD-POTFILES-t \
	    && chmod a-w BLD-POTFILES-t \
	    && mv BLD-POTFILES-t BLD-POTFILES )
cd .. \
	  && CONFIG_FILES=po/Makefile.in:po/Make-in \
	     CONFIG_HEADERS= /bin/bash ./config.status
config.status: creating po/Makefile.in
config.status: executing depfiles commands
config.status: executing libtool commands
config.status: executing default-1 commands
config.status: executing bfd_stdint.h commands
config.status: executing default commands
make[4]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd/po'
make[4]: Entering directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd/po'
make[4]: Nothing to be done for `info'.
make[4]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd/po'
make[4]: Entering directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd'
make[4]: Nothing to be done for `info-am'.
make[4]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd'
make[3]: *** [info-recursive] Error 1
make[3]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux/bfd'
make[2]: *** [all-bfd] Error 2
make[2]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/vagrant/source/risc-v/riscv-tools/riscv-gnu-toolchain/build-binutils-linux'
make: *** [stamps/build-binutils-linux] Error 2

awxhw, 2018-08-07:
@jim-wilson ,
 I‘m very appreciated for your answer. 
 The probem has been resoved by switching perl version to 5.18.2, as my ubuntu version is 14.04.5 LTS.

advaneharshal, 2022-11-22:
On the similar lines I am getting the following error on cygwin while execting makeinfo 
checking for makeinfo... makeinfo
Can't locate if.pm in @INC (you may need to install the if module) (@INC contains: /usr/lib/texinfo /usr/share/texinfo/lib/Text-Unidecode/lib /usr/share/texinfo/lib/Unicode-EastAsianWidth/lib /usr/share/texinfo/lib/libintl-perl/lib /usr/share/texinfo /usr/share/texinfo /usr/local/lib/perl5/site_perl/5.32/x86_64-cygwin-threads /usr/local/share/perl5/site_perl/5.32 /usr/lib/perl5/vendor_perl/5.32/x86_64-cygwin-threads /usr/share/perl5/vendor_perl/5.32 /usr/lib/perl5/5.32/x86_64-cygwin-threads /usr/share/perl5/5.32) at /usr/share/texinfo/Texinfo/Structuring.pm line 27.
BEGIN failed--compilation aborted at /usr/share/texinfo/Texinfo/Structuring.pm line 27.
Compilation failed in require at /usr/share/texinfo/Texinfo/Convert/Converter.pm line 32.
BEGIN failed--compilation aborted at /usr/share/texinfo/Texinfo/Convert/Converter.pm line 32.
Compilation failed in require at /usr/bin/makeinfo line 99.
BEGIN failed--compilation aborted at /usr/bin/makeinfo line 99.

I am using the latest version of perl 5.32.1, Any idea how to resolve this issue.