riscv-software-src/riscv-tools #230: error excuting  spike

awxhw, 2018-08-08:
when excuted a sample test by using command spike pk  test，but the error happened is as follows.
"
terminate called after throwing an instance of 'std::runtime_error'
  what():  couldn't allocate 2147483648 bytes of target memory
Aborted (core dumped)

"

And when excuted spike alone, the same error occurred, so I'm not sure about that.

heshamelmatary, 2018-08-08:
Looks like your host machine doesn't have enough memory. Spike tries to allocate 2 GiB by default. Try to explicitly run Spike with less memory, e.g.:

`spike -m128 pk test`

awxhw, 2018-08-09:
@jim-wilson 
    Thank you for your reply, I will use riscv-gnu-toolchain instead.

@heshamelmatary 
    Thanks, the problem has been resolved by running spike -m128 pk test, as the total memory of my virtual machine is only 2GiB.

awxhw, 2018-08-09:
@heshamelmatary 
when I run Spike to boot linux using the following command:
 $spike  -m256  bbl    riscv-tools/riscv-linux/vmlinux

Howerver, no boot log can be seen until I hit ctrl+C，the logs are as follow,
^C:
core   0: 0xffffffe00009137c (0x10016073) csrsi   sstatus, 2
:
core   0: 0xffffffe000091380 (0x00006422) ld      s0, 8(sp)
:
core   0: 0xffffffe000091382 (0x00000141) addi    sp, sp, 16
:
core   0: 0xffffffe000091384 (0x00008082) ret
:
core   0: 0xffffffe0000b3cde (0x100027f3) csrr    a5, sstatus
:
core   0: 0xffffffe0000b3ce2 (0x00008b89) andi    a5, a5, 2
:
core   0: 0xffffffe0000b3ce4 (0x0000c3a9) beqz    a5, pc + 66
:
core   0: 0xffffffe0000b3ce6 (0x432230ef) jal     pc + 0x23432
:
core   0: 0xffffffe0000d7118 (0x00001141) addi    sp, sp, -16
:
core   0: 0xffffffe0000d711a (0x0000e422) sd      s0, 8(sp)

Is there anything wrong with that?

heshamelmatary, 2018-08-09:
@awxhw  Sorry, I don't have experience running Linux on Spike, I only tried it recently on [QEMU](https://wiki.qemu.org/Documentation/Platforms/RISCV)

awxhw, 2018-08-09:
@heshamelmatary Ok， thanks.

jim-wilson, 2018-08-09:
sifive/freedom-u-sdk has working support to build a linux kernel+buildroot and boot it on qemu and spike.  Though the default serial driver doesn't work with spike, so you need to enable CONFIG_HVC_RISCV_SBI in conf/linux_defconfig to make that work.  It is just "make qemu" for a qemu boot and "make sim" for a spike boot.

closed 2018-10-26 by aswaterman